coolsnowwolf/lede #8070: 小米路由器4c 编译生成的固件不能用

pengdongmin, 2021-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：小米路由器4c，芯片是mt7628，之前用的都是别人分享lean大神的固件：openwrt-ramips-mt76x8-xiaomi_miwifi-4C-squashfs-sysupgrade.bin  10.2M，可以用。
后来我想自己编一个固件，按照大神的教程编译的，make menuconfig 后选择分别是 MediaTek Ralink MIPS、MT76x8 based boards、Xiaomi Mi Router 4C。编译成功后也生成了个sysupgrade.bin
然后我在luci页面固件升级失败了，显示“不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。”。然后我就直接在breed页面升级固件，升级完后路由器就橙灯和蓝灯一起闪，隔几秒闪一次，wifi也不见了。重新升级回别人的固件就又可以用了。。
我寻思着上面的型号也没选错呀....这是为什么呢？困扰了我好久，求助！

### (2) 路由器型号和固件版本

A：小米路由器4c

### (3) 详细日志

A：无

qinfeng8848, 2021-10-31:
我的小米路由4，我可以参考
先在breed页面升级固件，
openwrt-ramips-mt7621-xiaomi_mi-router-4-initramfs-kernel.bin

然后
1.
scp openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin  root@192.168.1.1:/tmp
mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin kernel

3.
mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin kernel_stock

2.
scp openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-rootfs0.bin root@192.168.1.1:/tmp
mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-rootfs0.bin ubi

pengdongmin, 2021-11-01:
> 我的小米路由4，我可以参考 先在breed页面升级固件， openwrt-ramips-mt7621-xiaomi_mi-router-4-initramfs-kernel.bin
> 
> 然后 1. scp openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin root@192.168.1.1:/tmp mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin kernel
> 
> 
> mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-kernel1.bin kernel_stock
> 
> 
> scp openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-rootfs0.bin root@192.168.1.1:/tmp mtd write openwrt-ramips-mt7621-xiaomi_mi-router-4-squashfs-rootfs0.bin ubi

谢谢你，不过我生成的固件里只有 initramfs-kernel.bin 和 squashfs-sysupgrade.bin ，没有 rootfs0.bin。。真不知道怎么办

hudi-ya, 2021-12-12:
好像是因为闪存布局的原因，坊间有办法，L大源码这里一直好像都还没解决

coolsnowwolf, 2026-09-06:
该问题是通过 Breed 刷写小米 4C 时的镜像格式/分区兼容问题，未提供完整分区表、构建配置或串口日志，且基于旧版本流程，无法确认当前 LEDE 镜像生成缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf