coolsnowwolf/lede #8071: qbittorrent 有较大下载任务时无法打开 Web UI

miyouzi, 2021-10-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

当下载任务较大时（8G以上大小，数十个文件），Web控制台（接口）无响应。其他安装的可能会影响的软件：`OpenClash` `Turbo ACC`

PS：固件的网页服务器已更换成 nginx

### (1) 具体问题

A：原生 Web UI 网页会直接超时，更换 [第三方UI](https://github.com/CzBiX/qb-web) 时网页可打开，但是数据无法加载。

通过负载监控可以发现其实有在下载。

[image]

[image]

### (2) 路由器型号和固件版本

A：小米路由3 Pro。2021-10-13  自编译版本
[image]

### (3) 详细日志

A：系统日志及内核日志均不包含 `qbittorrent` 关键字，内核无报错异常
[image]

hxlls, 2021-10-17:
7621就这样了，别要求太高了

coolsnowwolf, 2021-10-18:
性能问题

miyouzi, 2021-10-19:
> 是不是刷新完全没有数据，好像是这几版qbittorrent的问题，回退了到4.3.7的好像就不会了，我是j3455 12G内存，挂载任务多的时候，qbittorrent的webui卡死，但是还在工作😂

@hyy-666 确实我的qb版本是4.3.8，可以问下如何降级么

miyouzi, 2021-10-19:
在隔壁 qBittorrent 项目询问得到答复 qb 的 WebAPI 十分消耗性能，容易超时。虽然还不是很清楚为什么 CPU 没有满载的情况接口还是无响应，但应该没有除了更换设备的其他解了

closed 2021-10-19 by miyouzi