coolsnowwolf/lede #8074: git pull 最新 k2p uhttpd

yjd, 2021-10-17:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
连续测试好几次，创建新目录重新拉取代码，按步骤编译。breed刷到k2p后
第一次启动经常性web打不开，进ssh查看发现uhttpd进程没有启动。
如果你按背面的复位键就可能启动后uhttpd进程也没启动。
反复刷机偶尔能成功打开web后。如果你按下web界面的软重启也会导致uhttpd进程无法启动。
手动按参数启动会报错  I/O error
`/usr/sbin/uhttpd -f -h /www -r OpenWrt -x /cgi-bin -u /ubus -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80 -I cgi-bin/luci`

ps：uhttpd，uhttpd-mod-ubus 组件默认都有编译进去。

但是我重新创建新目录，拉取代码，回退到我原来一直用的 R21.4.18 就没问题
随便刷每次都能一次成功访问web，按后面复位键也没问题。
git reset --hard 420f085d950

### (2) 路由器型号和固件版本

A：k2p a2 官方原版、 R21.10.1 git pull 最新

### (3) 详细日志

A：web打不开就忘记抓日志了。提供我编译用的配置：
R21.4.18、R21.10.1
[配置文件.config.zip](https://github.com/coolsnowwolf/lede/files/7360099/config.zip)

nanchuci, 2021-10-19:
确实有这问题出现

yjd, 2021-10-30:
> 我是超频1000以上就打不开，900多能打开

我没有超频，估计你编译的是5.4.x内核的。之前我直接git clone下的最新是5.10.x内核，
没注意还有2个版本内核，改天再测试下5.4.x 最新内核

97xiami, 2021-10-30:
> > 我是超频1000以上就打不开，900多能打开
> 
> 我没有超频，估计你编译的是5.4.x内核的。之前我直接git clone下的最新是5.10.x内核，
> 没注意还有2个版本内核，改天再测试下5.4.x 最新内核

5.4.155，拉下来什么都不改都不行

yjd, 2021-10-30:
> 5.4.155，拉下来什么都不改都不行
感谢测试，那我就不用测试了。