coolsnowwolf/lede #8145: CR660x丢失网口连接

liwenjie119, 2021-10-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
网线连接wan口不停重连，lan口接电脑无链接
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：网线连接wan口，静态地址，连上wifi没有网，日志不停显示重连；而如果网线连lan口和电脑，电脑都没有显示连上，路由器日志就是显示lan口连上也立马掉线。

### (2) 路由器型号和固件版本

A：最新源码 d9a8c16
 编译用config

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-cr660x=y
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_autosamba=y
CONFIG_PACKAGE_ipv6helper=y
# CONFIG_PACKAGE_logd is not set
CONFIG_PACKAGE_syslog-ng=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-adguardhome=y
CONFIG_PACKAGE_luci-app-airplay2=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-frps=y
CONFIG_PACKAGE_luci-app-godproxy=y
CONFIG_PACKAGE_luci-app-guest-wifi=y
CONFIG_PACKAGE_luci-app-ipsec-vpnd=y
CONFIG_PACKAGE_luci-app-jd-dailybonus=y
CONFIG_PACKAGE_luci-app-minidlna=y
CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-mwan3helper=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-app-ntpc=y
CONFIG_PACKAGE_luci-app-oaf=y
CONFIG_PACKAGE_luci-app-pppoe-relay=y
CONFIG_PACKAGE_luci-app-serverchan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Libev_Client=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Libev_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Client=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust_Server=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Simple_Obfs=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray_Plugin=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Xray=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-uugamebooster=y
CONFIG_PACKAGE_luci-app-v2ray-server=y
CONFIG_PACKAGE_luci-app-webadmin=y
CONFIG_PACKAGE_luci-app-wireguard=y
CONFIG_PACKAGE_luci-app-wrtbwmon=y
CONFIG_PACKAGE_luci-app-zerotier=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_minieap=y
CONFIG_PACKAGE_openssh-sftp-avahi-service=y
CONFIG_PACKAGE_openssh-sftp-server=y

### (3) 详细日志
[... 1143 lines cut ...]
Oct 29 16:05:44 OpenWrt hostapd: wlan1: AP-STA-DISCONNECTED a4:b1:c1:c7:a1:8c
Oct 29 16:05:44 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: authenticated
Oct 29 16:05:44 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: associated (aid 1)
Oct 29 16:05:44 OpenWrt hostapd: wlan1: AP-STA-CONNECTED a4:b1:c1:c7:a1:8c
Oct 29 16:05:44 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c RADIUS: starting accounting session A50922846C014A5D
Oct 29 16:05:44 OpenWrt hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED a4:b1:c1:c7:a1:8c off
Oct 29 16:06:02 OpenWrt hostapd: wlan1: AP-STA-DISCONNECTED a4:b1:c1:c7:a1:8c
Oct 29 16:06:31 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: authenticated
Oct 29 16:06:31 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: associated (aid 1)
Oct 29 16:06:31 OpenWrt hostapd: wlan1: AP-STA-CONNECTED a4:b1:c1:c7:a1:8c
Oct 29 16:06:31 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c RADIUS: starting accounting session 78E1C046C66B5568
Oct 29 16:06:31 OpenWrt hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED a4:b1:c1:c7:a1:8c off
Oct 29 16:07:36 OpenWrt hostapd: wlan1: AP-STA-DISCONNECTED a4:b1:c1:c7:a1:8c
Oct 30 00:08:06 OpenWrt kernel: [  423.643704] mt7530 mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control rx/tx
Oct 30 00:08:06 OpenWrt kernel: [  423.651394] br-lan: port 1(lan1) entered blocking state
Oct 30 00:08:06 OpenWrt kernel: [  423.656646] br-lan: port 1(lan1) entered forwarding state
Oct 29 16:08:06 OpenWrt netifd: Network device 'lan1' link is up
Oct 29 16:08:17 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: authenticated
Oct 29 16:08:17 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c IEEE 802.11: associated (aid 1)
Oct 29 16:08:17 OpenWrt hostapd: wlan1: AP-STA-CONNECTED a4:b1:c1:c7:a1:8c
Oct 29 16:08:17 OpenWrt hostapd: wlan1: STA a4:b1:c1:c7:a1:8c RADIUS: starting accounting session 436FF5C9EE91C490
Oct 29 16:08:17 OpenWrt hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED a4:b1:c1:c7:a1:8c off
Oct 30 00:08:23 OpenWrt kernel: [  440.283477] mt7530 mdio-bus:1f lan1: Link is Down
Oct 30 00:08:23 OpenWrt kernel: [  440.288716] br-lan: port 1(lan1) entered disabled state
Oct 29 16:08:23 OpenWrt netifd: Network device 'lan1' link is down

liwenjie119, 2021-11-06:
测试了很多天，发现可能是与上级交换机的问题，学校实验室不知道用的啥交换机，最近的版本存在掉网口问题，这个交换机换不同口不同网线都不能使，拿到其他地方不是这个交换机就正常，特别奇怪。

coolsnowwolf, 2026-09-06:
CR660x 网口反复掉线报告没有完整内核/交换机日志，也未给出 PHY、网线和端口配置，无法确认是当前 ramips 驱动缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf