coolsnowwolf/lede #8146: 编译一直报错

Well2333, 2021-10-31:
(1) 具体问题
如题，编译一直报错，默认编译Ubuntu server 20.04没有成功过，Ubuntu desktop 18.04在默认编译下成功过一次，额外加一个插件（比如aria2）都会报错

(2) 路由器型号和固件版本
软路由，X86架构的，固件版本就是目前的最新版

(3) 详细日志
只把最后一次报错的放上了，之前的报错日志找不到了

`/home/well404/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src does not exist, skipping symlinks

touch /home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.configured_1cd727ac7b9ebbe4ee453b5ddb8504b3
rm -f /home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.built
touch /home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1=runc-1.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1=runc-1.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1=runc-1.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1=runc-1.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/well404/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.go_work/build" GOCACHE="/home/well404/lede/tmp/go-build" GOMODCACHE="/home/well404/lede/dl/go-mod-cache" GOENV=off make  -C /home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.go_work/build/src/github.com/opencontainers/runc AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1=runc-1.0.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" COMMIT=4144b63817ebcc5b358fc2c8ef95f7cddd709aa7 BUILDTAGS=' seccomp' ;
make[4]: Entering directory '/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.go_work/build/src/github.com/opencontainers/runc'
go build -trimpath "-mod=vendor" "-buildmode=pie"  -tags "seccomp" -ldflags "-X main.gitCommit=4144b63817ebcc5b358fc2c8ef95f7cddd709aa7 -X main.version=1.0.1 " -o runc .
# github.com/opencontainers/runc
/home/well404/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/link: running x86_64-openwrt-linux-musl-gcc failed: exit status 1
collect2: error: ld returned 1 exit status

Makefile:35: recipe for target 'runc' failed
make[4]: *** [runc] Error 2
make[4]: Leaving directory '/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.go_work/build/src/github.com/opencontainers/runc'
Makefile:64: recipe for target '/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.built' failed
make[3]: *** [/home/well404/lede/build_dir/target-x86_64_musl/runc-1.0.1/.built] Error 2
make[3]: Leaving directory '/home/well404/lede/feeds/packages/utils/runc'
time: package/feeds/packages/runc/compile#40.21#10.44#18.58
    ERROR: package/feeds/packages/runc failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/runc/compile' failed
make[2]: *** [package/feeds/packages/runc/compile] Error 1
make[2]: Leaving directory '/home/well404/lede'
package/Makefile:110: recipe for target '/home/well404/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/well404/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/well404/lede'
/home/well404/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2
well404@ubuntu:~/lede$ 
`

WYC-2020, 2021-10-31:
清理全新编译，我这边没问题

minlang112, 2021-10-31:
make distclean 可以破大部分奇怪的问题……

Well2333, 2021-11-01:
并不是，因为我是新配置的环境，也是首次编译，直接出的问题

closed 2021-11-12 by AmadeusGhost