coolsnowwolf/lede #8151: ipsec无法连接

blackshot-zard, 2021-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ipsec无法连接，默认编译，只增加了xx，有公网ip 无法连接

### (2) 路由器型号和固件版本

A：x86-64 j4125 2.5g

### (3) 详细日志

A：Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
Mon Nov  1 10:10:49 2021 daemon.info : 00[CFG]   loaded EAP secret for lean
Mon Nov  1 10:10:49 2021 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Mon Nov  1 10:10:49 2021 daemon.info : 00[JOB] spawning 16 worker threads
Mon Nov  1 10:10:49 2021 daemon.info : 05[DMN] thread 5 received 11
Mon Nov  1 10:10:49 2021 daemon.info : 06[DMN] thread 6 received 11
Mon Nov  1 10:10:49 2021 daemon.info : 06[LIB] no support for capturing backtraces
Mon Nov  1 10:10:49 2021 daemon.info : 06[DMN] killing ourself, received critical signal
Mon Nov  1 10:10:49 2021 authpriv.info ipsec_starter[8737]: charon has died -- restart scheduled (5sec)
Mon Nov  1 10:10:49 2021 authpriv.info ipsec_starter[8737]: charon refused to be started
Mon Nov  1 10:10:54 2021 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.9.3, Linux 5.4.155, x86_64)
Mon Nov  1 10:10:54 2021 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Mon Nov  1 10:10:54 2021 daemon.info : 00[KNL] failed to create TUN device
Mon Nov  1 10:10:54 2021 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Mon Nov  1 10:10:54 2021 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Mon Nov  1 10:10:54 2021 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
Mon Nov  1 10:10:54 2021 daemon.info : 00[CFG]   loaded EAP secret for lean
Mon Nov  1 10:10:54 2021 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Mon Nov  1 10:10:54 2021 daemon.info : 00[JOB] spawning 16 worker threads
Mon Nov  1 10:10:54 2021 daemon.info : 05[DMN] thread 5 received 11
Mon Nov  1 10:10:54 2021 daemon.info : 04[DMN] thread 4 received 11
Mon Nov  1 10:10:54 2021 daemon.info : 04[LIB] no support for capturing backtraces
Mon Nov  1 10:10:54 2021 daemon.info : 06[DMN] thread 6 received 11
Mon Nov  1 10:10:54 2021 daemon.info : 06[LIB] no support for capturing backtraces
Mon Nov  1 10:10:54 2021 daemon.info : 06[DMN] killing ourself, received critical signal
Mon Nov  1 10:10:54 2021 authpriv.info ipsec_starter[8737]: charon has died -- restart scheduled (5sec)
Mon Nov  1 10:10:54 2021 authpriv.info ipsec_starter[8737]: charon refused to be started
Mon Nov  1 10:10:59 2021 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.9.3, Linux 5.4.155, x86_64)
Mon Nov  1 10:10:59 2021 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Mon Nov  1 10:10:59 2021 daemon.info : 00[KNL] failed to create TUN device
Mon Nov  1 10:10:59 2021 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Mon Nov  1 10:10:59 2021 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Mon Nov  1 10:10:59 2021 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Mon Nov  1 10:10:59 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
[... 643 lines cut ...]
Mon Nov  1 10:13:19 2021 authpriv.info ipsec_starter[8737]: charon refused to be started
Mon Nov  1 10:13:24 2021 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.9.3, Linux 5.4.155, x86_64)
Mon Nov  1 10:13:24 2021 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Mon Nov  1 10:13:24 2021 daemon.info : 00[KNL] failed to create TUN device
Mon Nov  1 10:13:24 2021 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Mon Nov  1 10:13:24 2021 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Mon Nov  1 10:13:24 2021 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
Mon Nov  1 10:13:24 2021 daemon.info : 00[CFG]   loaded EAP secret for lean
Mon Nov  1 10:13:24 2021 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Mon Nov  1 10:13:24 2021 daemon.info : 00[JOB] spawning 16 worker threads
Mon Nov  1 10:13:24 2021 daemon.info : 05[DMN] thread 5 received 11
Mon Nov  1 10:13:24 2021 daemon.info : 05[LIB] no support for capturing backtraces
Mon Nov  1 10:13:24 2021 daemon.info : 06[DMN] thread 6 received 11
Mon Nov  1 10:13:24 2021 daemon.info : 06[LIB] no support for capturing backtraces
Mon Nov  1 10:13:24 2021 daemon.info : 06[DMN] killing ourself, received critical signal
Mon Nov  1 10:13:24 2021 daemon.info : 05[DMN] killing ourself, received critical signal
Mon Nov  1 10:13:24 2021 authpriv.info ipsec_starter[8737]: charon has died -- restart scheduled (5sec)
Mon Nov  1 10:13:24 2021 authpriv.info ipsec_starter[8737]: charon refused to be started

deplives, 2021-11-01:
没有编译对应的tun内核模块

blackshot-zard, 2021-11-01:
> 没有编译对应的tun内核模块

我搜谷歌找到了这个问题，但是加了之后编译不通过。。。

blackshot-zard, 2021-11-02:
> 没有编译对应的tun内核模块

全部重新设置之后，加了kmod tun模块后成功了 谢谢

closed 2021-11-02 by blackshot-zard