coolsnowwolf/lede #8152: ax3600编译vlmcsd报错

zcluo, 2021-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译VLMCSD报错

### (2) 路由器型号和固件版本

A：ax3600 

### (3) 详细日志

A：rm -rf /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd.installed /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd
mkdir -p /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd
install -d -m0755 /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd/usr/bin
install -m0755 /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/bin/vlmcsd /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd/usr/bin/vlmcsd
install -d -m0755 /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd/etc
install -m0755 ./files/vlmcsd.ini /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd/etc/vlmcsd/vlmcsd.ini
install: cannot create regular file '/home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd/etc/vlmcsd/vlmcsd.ini': No such file or directory
make[3]: *** [Makefile:46: /home/zcluo/lede/build_dir/target-aarch64_cortex-a53_musl/vlmcsd-svn1113/.pkgdir/vlmcsd.installed] Error 1
make[3]: Leaving directory '/home/zcluo/lede/package/lean/vlmcsd'
time: package/lean/vlmcsd/compile#0.14#0.02#0.15
    ERROR: package/lean/vlmcsd failed to build.
make[2]: *** [package/Makefile:116: package/lean/vlmcsd/compile] Error 1
make[2]: Leaving directory '/home/zcluo/lede'

lxwq, 2021-11-01:
X86也一样报错

ZqinKing, 2021-11-01:
修改 package/lean/vlmcsd/Makefile文件
$(INSTALL_BIN) ./files/vlmcsd.ini $(1)/etc/vlmcsd/vlmcsd.ini ==> $(INSTALL_BIN) ./files/vlmcsd.ini $(1)/etc/vlmcsd.ini

AmadeusGhost, 2021-11-01:
已修复

closed 2021-11-01 by AmadeusGhost