coolsnowwolf/lede #8157: N1盒子编译出来

siren202101, 2021-11-01:
[image]
新源码编译出来的openwrt-armvirt-64-default-rootfs
要么找不到install-to-emmc.sh,要么运行不了install-to-emmc.sh，运行了又变成这样是什么原理？如何解决？

ZqinKing, 2021-11-01:
大兄弟，install-to-emmc.sh不是这个项目里带的吧。

ZqinKing, 2021-11-01:
https://github.com/unifreq/openwrt_packit.git

siren202101, 2021-11-01:
> https://github.com/unifreq/openwrt_packit.git

   这个有点玄学，我试用使用openwrt_packit.git前两个版本的来打包也是失败
openwrt_packit.git打包旧源码编译的openwrt-armvirt-64-default-rootfs成功，打包新源码编译的openwrt-armvirt-64-default-rootfs失败
就是不知道是那边出的问题所以都问问看

yonggedebaqi, 2021-11-01:
提示 缺什么补什么再编译

kissyouhunter, 2021-11-01:
> > https://github.com/unifreq/openwrt_packit.git
> 
> 这个有点玄学，我试用使用openwrt_packit.git前两个版本的来打包也是失败 openwrt_packit.git打包旧源码编译的openwrt-armvirt-64-default-rootfs成功，打包新源码编译的openwrt-armvirt-64-default-rootfs失败 就是不知道是那边出的问题所以都问问看

F大新版的打包没问问题的
三个内核都正常

closed 2021-11-02 by AmadeusGhost