coolsnowwolf/lede #8159: "状态/概览"页面-网络信息显示不正确

felixxiao99, 2021-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：将R2S用作旁路由，删除了WAN、WAN6，只保留LAN，并将LAN对应的网卡由预设的eth1改为eth0，使用上没有问题，但在“状态/概览”页面的网络信息显示不正确

### (2) 路由器型号和固件版本

A：
路由器型号：NanoPi R2S 
固件版本：OpenWrt R21.11.11 Compiled by abc / LuCI Master (git-21.280.14233-9b796b1)

### (3) 详细日志

[image]

[image]

WYC-2020, 2021-11-08:
确实是这个问题，lan口直接显示在wan状态那栏 ，在没有wan口的时候

WYC-2020, 2021-11-08:
通过从恩山上面了解到 好像你把网关设置成和某一个端口的ip在同一网段，他就认为这个端口是wan口，你10.9.9.2和10.9.9.1 是在同一个网段他就会认为你这个是wan口，改成不同网段即可

coolsnowwolf, 2026-09-06:
当前 LuCI 概览页已使用新版网络状态模型，支持仅 LAN、非桥接物理接口及 WireGuard 等接口显示；旧版概览分类错误问题已被后续 LuCI 重写覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf