coolsnowwolf/lede #8160: mt7621莫名重启或死机，请帮忙看下日志

bonnyandsky, 2021-11-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
10月21日编译了r3823-13384d4c5后在使用过程中发现每天早上6-9点会随机重启设备。
然后我在11月1日重新编译了r3844-050941517进行刷机，不保留原配制，刷机后所有配制手动设置，未导入任何之前的配置备份。然后今天早上7点多醒来发现不能上网，路由器处于死机状态，路由器的dchp服务器不工作，路由器ip也ping不通，只能手动重启。
### (1) 具体问题

A：请帮忙看看日志，什么原因导致死机或者重启？

### (2) 路由器型号和固件版本

A：xianyu c5  r3844-050941517

### (3) 详细日志

 

> #
[All_2021-11-2-7_36_56.csv](https://github.com/coolsnowwolf/lede/files/7456720/All_2021-11-2-7_36_56.csv)

bonnyandsky, 2021-11-02:
应为看到有大量watchcat的错误信息
[image]
于是我去看了下watchcat的配置文件，发现默认是有相关设置的，但是我web设置界面里没有设置过。而且我尝试在web设置界面添加设置并保存后，watchcat的配置文件内容也不会有对应变化。
[image]
另外日志里面有一段时间顺序有问题，不知道是什么原因造成的。
[image]

[image]

bonnyandsky, 2021-11-02:
然后我发现日志里还有很多cpu停顿的信息。看不太懂，也不知道是什么情况。感觉今天早上的司机貌似和watchcat没关系的？
[image]

tmpaxx, 2021-11-09:
摸摸cpu 是不是非常烫？7621不行了

bonnyandsky, 2021-11-09:
> 摸摸cpu 是不是非常烫？7621不行了

你是说硬件导致的？我开了off必死，关闭的话隔几天死一次。这几天死机就去看日志，并没有任何记录。

tmpaxx, 2021-11-10:
你看CPU占用率，是不是非常高，死机了没有日志的。性能不行了

bonnyandsky, 2021-11-11:
> 你看CPU占用率，是不是非常高，死机了没有日志的。性能不行了

我知道日志在内存里会丢失，所以我日志都放到群晖上了方便查看，但是我发现死机前也没日志，只有之前正常的那些日志内容。
而且最近2次死机家里也没人用网络。只有群晖开着PT在上传，带宽量也很小。刚远程看了下cpu占用率也不高

平均负载 | 0.15, 0.09, 0.02

ddd-zero, 2022-02-16:
问一下怎么保存系统日志和内核日志到nas上？
我也是7621 开启硬件后 一天就死机重启。不开可以稳定半个月。
死机前负载会到1。而关闭硬件后，日常上网只有0.1

yuanshus, 2022-10-19:
> 
那要如何关闭硬件

yuanshus, 2022-10-19:
> 问一下怎么保存系统日志和内核日志到nas上？ 我也是7621 开启硬件后 一天就死机重启。不开可以稳定半个月。 死机前负载会到1。而关闭硬件后，日常上网只有0.1

请问如何关闭硬件？我的7621也遇到了这个问题

closed 2026-09-05 by coolsnowwolf