coolsnowwolf/lede #8165: bcm53xx   k3编译5.10.76内核固件报错

392377870, 2021-11-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：bcm53xx   K3 编译5.10.76内核固件

### (3) 详细日志

A：rm -f /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac56u-kernel.bin
rm -f /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac68u-kernel.bin
rm -f /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac87u-kernel.bin
cp /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac56u-kernel.bin
rm -f /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-n18u-kernel.bin
Parallel mksquashfs: Using 8 processors
Creating 4.0 filesystem on /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/root.squashfs, block size 1048576.
cp /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac68u-kernel.bin
cp /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac87u-kernel.bin
cp /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/zImage /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-n18u-kernel.bin
cat /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb >> /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac56u-kernel.bin
cat /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb >> /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac68u-kernel.bin
cat /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb >> /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac87u-kernel.bin
cat /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb >> /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-n18u-kernel.bin
cat: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb: No such file or directory
cat: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb: No such file or directory
cat: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb: No such file or directory
make[5]: *** [Makefile:455: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac87u-kernel.bin] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [Makefile:455: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac68u-kernel.bin] Error 1
cat: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/image-.dtb: No such file or directory
make[5]: *** [Makefile:455: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-ac56u-kernel.bin] Error 1
make[5]: *** [Makefile:455: /home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/asus_rt-n18u-kernel.bin] Error 1
Pseudo file "/dev" exists in source filesystem "/home/openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/dev".
Ignoring, exclude it (-e/-ef) to override.
[========================================================================================================|] 1698/1698 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 1048576
	compressed data, compressed metadata, compressed fragments,
	no xattrs, compressed ids
	duplicates are removed
Filesystem size 16761.82 Kbytes (16.37 Mbytes)
	33.82% of uncompressed filesystem size (49556.28 Kbytes)
Inode table size 13994 bytes (13.67 Kbytes)
	20.36% of uncompressed inode table size (68729 bytes)
Directory table size 20220 bytes (19.75 Kbytes)
	42.67% of uncompressed directory table size (47387 bytes)
Number of duplicate files found 268
Number of inodes 2092
Number of files 1694
Number of fragments 20
Number of symbolic links  235
Number of device nodes 1
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 162
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
make[5]: Leaving directory '/home/openwrt/lede/target/linux/bcm53xx/image'
make[4]: *** [Makefile:30: install] Error 2
make[4]: Leaving directory '/home/openwrt/lede/target/linux/bcm53xx'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/openwrt/lede/target/linux'
time: target/linux/install#88.00#8.75#26.35
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [target/Makefile:19: /home/openwrt/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:230：world] 错误 2
openwrt@ubuntu:~/lede$

laolarou726, 2021-11-03:
我这边也遇到了相同的问题，同时在执行```make -j8 download V=s```下载DL库时，最后一个依赖始终下载不下来，提示超时。

coolsnowwolf, 2026-09-06:
提供的 bcm53xx/K3 编译输出被截断在镜像生成阶段，没有实际失败行、配置或可复现环境，无法定位当前构建系统缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf