coolsnowwolf/lede #8166: 当lan口不使用桥接，直接使用物理接口时， luci获取不到lan口接口状态数据

hicaoc, 2021-11-03:
## 2. 详细叙述

LAN口关闭桥接，直接使用物理接口时，luci无法获取到接口信息

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

Intel 4125 
 OpenWrt R21.10.1 / LuCI Master (git-21.280.14233-9b796b1)
Intel Core Processor (Haswell, no TSX, IBRS) : 1 Core 2 Thread (CpuMark : 40123.044002 Scores)

更老的版本也测试过， 都有这个问题，必须使用br-lan 才能正常显示所有接口的信息

另外，在没有wan口的时候（单LAN）的时后，概述部分的LAN接口显示在IP WAN下面
<html>
<body>

Pv4 WAN 状态 | br-lan类型: static地址: 192.168.0.2子网掩码: 255.255.255.0网关: 192.168.0.1DNS 1: 114.114.114.114DNS 2: 218.2.135.1已连接: 15h 56m 30s | br-lan | 类型: static地址: 192.168.0.2子网掩码: 255.255.255.0网关: 192.168.0.1DNS 1: 114.114.114.114DNS 2: 218.2.135.1已连接: 15h 56m 30s
-- | -- | -- | --
br-lan | 类型: static地址: 192.168.0.2子网掩码: 255.255.255.0网关: 192.168.0.1DNS 1: 114.114.114.114DNS 2: 218.2.135.1已连接: 15h 56m 30s

</body>
</html>

A：

### (3) 详细日志
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: The called action terminated with an exception:
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: stack traceback:
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: 	[C]: in function 'assert'
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Nov  2 16:09:09 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Nov  2 16:09:10 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Nov  2 16:09:10 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Nov  2 16:09:10 2021 cron.err crond[14027]: crond (busybox 1.33.1) started, log level 8
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: The called action terminated with an exception:
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: stack traceback:
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: 	[C]: in function 'assert'
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Nov  2 16:09:35 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: The called action terminated with an exception:
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: stack traceback:
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: 	[C]: in function 'assert'
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Nov  2 16:09:38 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: The called action terminated with an exception:
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: stack traceback:
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: 	[C]: in function 'assert'
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Nov  2 16:09:43 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/network/iface_status/lan'.
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: The called action terminated with an exception:
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: /usr/lib/lua/luci/model/network.lua:930: bad argument #2 to '?' (no value)
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: stack traceback:
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: 	[C]: in function 'assert'
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Nov  2 16:09:44 2021 daemon.err uhttpd[2612]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
A：

felixxiao99, 2021-11-03:
遇到同样问题[https://github.com/coolsnowwolf/lede/issues/8159](url)

XiaoliChan, 2021-11-05:
What did you mean?

WYC-2020, 2021-11-08:
重启网络

WYC-2020, 2021-11-08:
这也算是一个bug 是你去掉桥接后在获取vlan的时候有什么获取不到，结果代码没有判断返回值，导致异常了，所以这个可以完善一下

hicaoc, 2021-11-18:
@WYC-2020 是的，虽然重启网络能恢复正常，但这个代码还是可以优化下的，代码本身应该能正确处理这个返回值，而不是直接异常退出

coolsnowwolf, 2026-09-06:
当前 LuCI 已改用新版 JavaScript 网络状态页面，接口状态获取不再依赖旧 Lua network.lua 中对桥接设备的假设；该 issue 针对 R21.10 的异常已被现行 LuCI 实现覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf