coolsnowwolf/lede #8173: 编译报错：No rule to make target 'compat/column.in', needed by 'po/quilt.pot'

zjjxwhh, 2021-11-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

将代码 clone 到本地后，feed 中添加 helloworld 及 passwall，按照 README 流程编译报错

### (2) 路由器型号和固件版本

x86_64

### (3) 详细日志

```
make[3]: Entering directory '/home/user/openwrt/tools/quilt'
. /home/user/openwrt/include/shell.sh; gzip -dc /home/user/openwrt/dl/quilt-0.66.tar.gz | tar -C /home/user/openwrt/build_dir/host/quilt-0.66/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/user/openwrt/build_dir/host/quilt-0.66

Applying ./patches/000-relocatable.patch using plaintext:
patching file bin/quilt.in
patching file quilt/scripts/edmail.in
patching file quilt/scripts/patchfns.in
patching file quilt/scripts/remove-trailing-ws.in
patching file Makefile.in
patching file quilt/scripts/backup-files.in
patching file bin/guards.in
patching file compat/date.in
patching file compat/getopt.in
patching file quilt/scripts/dependency-graph.in

Applying ./patches/001-fix_compile.patch using plaintext:
patching file Makefile.in
touch /home/user/openwrt/build_dir/host/quilt-0.66/.prepared703b89c3419b858cc343897a049852e0_6664517399ebbbc92a37c5bb081b5c53
cd /home/user/openwrt/build_dir/host/quilt-0.66 && autoconf
(cd /home/user/openwrt/build_dir/host/quilt-0.66/; if [ -x configure ]; then cp -fpR /home/user/openwrt/scripts/config.{guess,sub} /home/user/openwrt/build_dir/host/quilt-0.66// && CC="gcc" CFLAGS="-O2 -I/home/user/openwrt/staging_dir/host/include " CXX="g++" CPPFLAGS="-I/home/user/openwrt/staging_dir/host/include " LDFLAGS="-L/home/user/openwrt/staging_dir/host/lib " CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/user/openwrt/staging_dir/host --exec-prefix=/home/user/openwrt/staging_dir/host --sysconfdir=/home/user/openwrt/staging_dir/host/etc --localstatedir=/home/user/openwrt/staging_dir/host/var --sbindir=/home/user/openwrt/staging_dir/host/bin ; fi )
checking for a BSD-compatible install... /home/user/openwrt/staging_dir/host/bin/install -c
checking whether #! works in shell scripts... yes
checking for bash... /home/user/openwrt/staging_dir/host/bin/bash
checking whether /home/user/openwrt/staging_dir/host/bin/bash quoting works... yes
checking for gcp... no
checking for cp... /home/user/openwrt/staging_dir/host/bin/cp
checking for gdate... no
checking for date... /usr/bin/date
checking whether /usr/bin/date --rfc-822 works... yes
checking for perl... /home/user/openwrt/staging_dir/host/bin/perl
checking for grep... /home/user/openwrt/staging_dir/host/bin/grep
checking whether /home/user/openwrt/staging_dir/host/bin/grep -q works... yes
checking whether /home/user/openwrt/staging_dir/host/bin/grep understands (foo|bar)... yes
checking for tail... /usr/bin/tail
checking whether /usr/bin/tail -n works... yes
checking for tr... /usr/bin/tr
checking whether /usr/bin/tr understands a-z ... yes
checking for stat... /home/user/openwrt/staging_dir/host/bin/stat
checking whether /home/user/openwrt/staging_dir/host/bin/stat -c '%h' works... yes
checking for sed... /home/user/openwrt/staging_dir/host/bin/sed
checking whether /home/user/openwrt/staging_dir/host/bin/sed understands (foo|bar)... yes
checking for gawk... /usr/bin/gawk
checking whether /usr/bin/gawk supports sub... yes
checking for pod2man... /usr/bin/pod2man
checking for column... no
configure: WARNING: Using internal column mechanism.  Use option '--with-column' to override
checking for getopt... /home/user/openwrt/staging_dir/host/bin/getopt
checking for getopt --long syntax... yes
checking for mktemp... /usr/bin/mktemp
checking whether /usr/bin/mktemp -d works... yes
checking for diff... /home/user/openwrt/staging_dir/host/bin/diff
checking for patch... /home/user/openwrt/staging_dir/host/bin/patch
checking the version of /home/user/openwrt/staging_dir/host/bin/diff... 3.7
checking the version of /home/user/openwrt/staging_dir/host/bin/patch... 2.7.6
checking for find... /home/user/openwrt/staging_dir/host/bin/find
checking whether /home/user/openwrt/staging_dir/host/bin/find -path works... yes
checking whether /home/user/openwrt/staging_dir/host/bin/find -print0 works... yes
checking for xargs... /home/user/openwrt/staging_dir/host/bin/xargs
checking whether /home/user/openwrt/staging_dir/host/bin/xargs -0 works... yes
checking for diffstat... no
configure: WARNING: diffstat not found, some optional functionalities will be missing
checking for sendmail... no
configure: WARNING: Using internal sendmail mechanism.  Use option '--with-sendmail' to override
checking for msgmerge... /usr/bin/msgmerge
checking for msgfmt... /usr/bin/msgfmt
checking for xgettext... /usr/bin/xgettext
checking for msguniq... /usr/bin/msguniq
checking for msgcat... /usr/bin/msgcat
checking for rpmbuild... no
checking for rpm... no
configure: WARNING: rpmbuild not found, some optional functionalities will be missing
checking for gmd5sum... no
checking for md5sum... /usr/bin/md5sum
checking for 7zr... /usr/bin/7zr
configure: creating ./config.status
config.status: creating Makefile

quilt version 0.66 configured.

Using '/home/user/openwrt/staging_dir/host' for installation prefix.

Report bugs to quilt-dev@nongnu.org
[ -f /home/user/openwrt/build_dir/host/quilt-0.66/Makefile ]
touch /home/user/openwrt/build_dir/host/quilt-0.66/.configured
make -C /home/user/openwrt/build_dir/host/quilt-0.66 SHELL="bash" all
make[4]: Entering directory '/home/user/openwrt/build_dir/host/quilt-0.66'
bin/quilt.in -> bin/quilt
bin/guards.in -> bin/guards
quilt/upgrade.in -> quilt/upgrade
quilt/refresh.in -> quilt/refresh
quilt/fold.in -> quilt/fold
quilt/revert.in -> quilt/revert
quilt/header.in -> quilt/header
quilt/remove.in -> quilt/remove
quilt/top.in -> quilt/top
quilt/applied.in -> quilt/applied
quilt/graph.in -> quilt/graph
quilt/edit.in -> quilt/edit
quilt/annotate.in -> quilt/annotate
quilt/mail.in -> quilt/mail
quilt/diff.in -> quilt/diff
quilt/add.in -> quilt/add
quilt/push.in -> quilt/push
quilt/pop.in -> quilt/pop
quilt/new.in -> quilt/new
quilt/import.in -> quilt/import
quilt/previous.in -> quilt/previous
quilt/grep.in -> quilt/grep
quilt/fork.in -> quilt/fork
quilt/delete.in -> quilt/delete
quilt/patches.in -> quilt/patches
quilt/snapshot.in -> quilt/snapshot
quilt/files.in -> quilt/files
quilt/series.in -> quilt/series
quilt/setup.in -> quilt/setup
quilt/rename.in -> quilt/rename
quilt/next.in -> quilt/next
quilt/unapplied.in -> quilt/unapplied
quilt/scripts/patchfns.in -> quilt/scripts/patchfns
quilt/scripts/inspect-wrapper.in -> quilt/scripts/inspect-wrapper
quilt/scripts/dependency-graph.in -> quilt/scripts/dependency-graph
quilt/scripts/edmail.in -> quilt/scripts/edmail
quilt/scripts/remove-trailing-ws.in -> quilt/scripts/remove-trailing-ws
quilt/scripts/backup-files.in -> quilt/scripts/backup-files
rm -f compat/column.in compat/mktemp.in compat/sendmail.in compat/date.in compat/getopt.in
README.in -> README
quilt.1.in -> quilt.1
/usr/bin/pod2man bin/guards > doc/guards.1
make[4]: *** No rule to make target 'compat/column.in', needed by 'po/quilt.pot'.  Stop.
make[4]: Leaving directory '/home/user/openwrt/build_dir/host/quilt-0.66'
make[3]: *** [Makefile:37: /home/user/openwrt/build_dir/host/quilt-0.66/.built] Error 2
make[3]: Leaving directory '/home/user/openwrt/tools/quilt'
time: tools/quilt/compile#0.07#0.03#1.16
    ERROR: tools/quilt failed to build.
make[2]: *** [tools/Makefile:159: tools/quilt/compile] Error 1
make[2]: Leaving directory '/home/user/openwrt'
make[1]: *** [tools/Makefile:155: /home/user/openwrt/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/user/openwrt'
make: *** [/home/user/openwrt/include/toplevel.mk:230: world] Error 2
```

closed 2022-02-14 by zjjxwhh

tiandan6666, 2023-08-30:
how did you solve the problem?

lliioollcn, 2024-04-01:
> It seems that make runs target compat (which deletes the file "compat/column.in") before target po/quilt.pot. This order must be reversed. I fix it in the file tools/quilt/Makefile, replacing the line: $(MAKE) -C $(HOST_BUILD_DIR) SHELL="$(BASH)" all with: $(MAKE) -C $(HOST_BUILD_DIR) SHELL="$(BASH)" po/quilt.pot all 

https://forum.openwrt.org/t/tools-quilt-compile-failed/78729