coolsnowwolf/lede #8174: 固件编译成功！从breed写入固件ok！但是更改初始密码 WiFi名称 加密协议及lan地址在路由器重新启动后失效！恢复到初始设置！

jeasonliang, 2021-11-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
固件编译成功！从breed写入固件ok！但是更改初始密码 WiFi名称 加密协议及lan地址在路由器重新启动后失效！恢复到初始设置！
### (1) 具体问题先前设置的登录密码！WiFi名称加密协议以及lan地址都每一步都点击保存及应用！但是断电重新启动失效！
无法保存系统设置！
A：

### (2) 路由器型号和固件版本
小米4A千兆版本

### (3) 详细日志
下面是.config详细内容 make menuconfig生产的：

#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y

CONFIG_TARGET_ramips=y

CONFIG_TARGET_ramips_mt7621=y

CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-4a-gigabit=y

CONFIG_HAS_SUBTARGETS=y
CONFIG_HAS_DEVICES=y
CONFIG_TARGET_BOARD="ramips"
CONFIG_TARGET_SUBTARGET="mt7621"
CONFIG_TARGET_PROFILE="DEVICE_xiaomi_mi-router-4a-gigabit"
CONFIG_TARGET_ARCH_PACKAGES="mipsel_24kc"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc"
CONFIG_CPU_TYPE="24kc"
CONFIG_LINUX_5_4=y
CONFIG_DEFAULT_base-files=y
CONFIG_DEFAULT_block-mount=y
CONFIG_DEFAULT_busybox=y
CONFIG_DEFAULT_coremark=y
CONFIG_DEFAULT_ddns-scripts_aliyun=y
CONFIG_DEFAULT_ddns-scripts_dnspod=y
CONFIG_DEFAULT_default-settings=y
CONFIG_DEFAULT_dnsmasq-full=y
CONFIG_DEFAULT_dropbear=y
CONFIG_DEFAULT_firewall=y
CONFIG_DEFAULT_fstools=y
CONFIG_DEFAULT_iptables=y
CONFIG_DEFAULT_iwinfo=y
CONFIG_DEFAULT_kmod-crypto-hw-eip93=y
CONFIG_DEFAULT_kmod-gpio-button-hotplug=y
CONFIG_DEFAULT_kmod-ipt-raw=y
CONFIG_DEFAULT_kmod-leds-gpio=y
CONFIG_DEFAULT_kmod-mt7603=y
CONFIG_DEFAULT_kmod-mt76x2=y
CONFIG_DEFAULT_kmod-nf-nathelper=y
CONFIG_DEFAULT_kmod-nf-nathelper-extra=y
CONFIG_DEFAULT_libc=y
CONFIG_DEFAULT_libgcc=y
CONFIG_DEFAULT_libustream-openssl=y
CONFIG_DEFAULT_logd=y
CONFIG_DEFAULT_luci=y
CONFIG_DEFAULT_luci-app-accesscontrol=y
CONFIG_DEFAULT_luci-app-arpbind=y
CONFIG_DEFAULT_luci-app-autoreboot=y
[... 4913 lines cut ...]

#
# NTRIP
#
# CONFIG_PACKAGE_ntripcaster is not set
# CONFIG_PACKAGE_ntripclient is not set
# CONFIG_PACKAGE_ntripserver is not set
# end of NTRIP

#
# OLSR.org network framework
#
# CONFIG_PACKAGE_oonf-dlep-proxy is not set
# CONFIG_PACKAGE_oonf-dlep-radio is not set
# CONFIG_PACKAGE_oonf-init-scripts is not set
# CONFIG_PACKAGE_oonf-olsrd2 is not set
# end of OLSR.org network framework

#
# Open vSwitch
#
# CONFIG_PACKAGE_openvswitch is not set
# CONFIG_PACKAGE_openvswitch-ovn-host is not set
# CONFIG_PACKAGE_openvswitch-ovn-north is not set
# CONFIG_PACKAGE_openvswitch-python3 is not

97xiami, 2021-11-04:
你刷的ramdisk文件吧，这个重启后就会恢复默认的
你要刷squashfs的那个才行

jeasonliang, 2021-11-10:
openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin
openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest
openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
这三个文件 在breed下刷入sysupgrade无法刷入。。。

jeasonliang, 2021-11-10:
难道是 breed版问题？我是这里下载的
https://breed.hackpascal.net/

97xiami, 2021-11-10:
> openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin
> openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest
> openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
> 这三个文件 在breed下刷入sysupgrade无法刷入。。。 

无法刷入是什么意思？报错还是无法开机？

jeasonliang, 2021-11-10:
> > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin
> > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest
> > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
> > 这三个文件 在breed下刷入sysupgrade无法刷入。。。
> 
> 无法刷入是什么意思？报错还是无法开机？

首先感谢你的回复！
无法开机  黄灯不停闪烁！

97xiami, 2021-11-10:
> > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin
> > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest
> > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
> > > 这三个文件 在breed下刷入sysupgrade无法刷入。。。
> > 
> > 无法刷入是什么意思？报错还是无法开机？
> 
> 首先感谢你的回复！
> 无法开机  黄灯不停闪烁！

换个分区刷

jeasonliang, 2021-11-10:
> > > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-initramfs-kernel.bin
> > > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit.manifest
> > > > openwrt-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin
> > > > 这三个文件 在breed下刷入sysupgrade无法刷入。。。
> > > 
> > > 
> > > 无法刷入是什么意思？报错还是无法开机？
> > 
> > 
> > 首先感谢你的回复！
> > 无法开机  黄灯不停闪烁！
> 
> 换个分区刷

估计是这个https://www.right.com.cn/FORUM/thread-4052254-1-1.html

closed 2021-11-12 by AmadeusGhost