coolsnowwolf/lede #8179: R7800 5.10内核固件编译失败，求帮助

hubbylei, 2021-11-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译R7800为什么会用到wg2600hp3的dts?

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
make[5]: Nothing to be done for 'Image'.
  DTC     arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb
Error: arch/arm/boot/dts/qcom-ipq8062.dtsi:97.1-6 Label or path nss0 not found
Error: arch/arm/boot/dts/qcom-ipq8062.dtsi:103.1-6 Label or path nss1 not found
FATAL ERROR: Syntax error parsing input tree
make[6]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb] Error 1
make[5]: *** [Makefile:1359: dtbs] Error 2
make[5]: Leaving directory '/home/hubbylei/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.10.77'
make[4]: *** [Makefile:29: /home/hubbylei/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.10.77/.modules] Error 2
make[4]: Leaving directory '/home/hubbylei/lede/target/linux/ipq806x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/hubbylei/lede/target/linux'
time: target/linux/compile#1622.59#228.06#1846.80
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/hubbylei/lede'
make[1]: *** [target/Makefile:18: /home/hubbylei/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/hubbylei/lede'
make: *** [/home/hubbylei/lede/include/toplevel.mk:230: world] Error 2

AmadeusGhost, 2021-11-08:
不支持 5.10，自行回退吧

hubbylei, 2021-11-08:
> 不支持 5.10，自行回退吧

好的，谢谢 :)

closed 2021-11-12 by AmadeusGhost