coolsnowwolf/lede #8187: 云编译一直没成功，最近资源被封的厉害，本地编译的几个注意点，像我这样的编译新手可能会踩的坑，供参考

alexgzhou, 2021-11-05:
1. 建议用国外云服务器编译，本地就没尝试了，网络肯定是个大坑，有梯子也没用，网络不稳定就要老命；
2. 编译对内存有要求，最好大一点，可用内存至少 `1GB` 以上，不够的话就加 `swapfile` 虚拟内存吧，增加的步骤网上一查就有，编译到某个步骤卡住了，然后报错基本都是这个原因（如果网络没问题的话）；
3. `$ make menuconfig` 对终端显示大小竟然有要求，这个有点坑，你把 `ssh` 终端拉大一点就好了，会有提示，起码 `19*81` 大小，否则会出现打不开 `menu` 自动跳过 `config` 步骤这样的问题，你人走开的话可能根本发现不了，更奇怪的是，如果窗口不够大，有时候进去 `menu` 了，但是竟然无法回车进入 `Target System` 等前三个子选项选目标芯片等操作，下面的其他选项可以进入，这种问题也太奇怪了，一般人根本想不到是哪里出问题了，我查了半天也没找到其他人碰到过。。。
4. 你用 `ssh` 登陆远程服务器编译的话，你得一直连着 `ssh`，否则网络问题断开后编译就中断了（国外服务器容易出现 `ssh` 间歇性断开的问题），可以 `nohup + &` 改造一下 `make -j$(($(nproc) + 1)) V=s` 命令。
5. 希望大家少走弯路，编译愉快！

10000plus, 2021-11-05:
云编译怎么不试试用github action

alexgzhou, 2021-11-05:
用了啊，我的一个主账号开始没注意，同时编译了两个actions，然后被封了，现在还没解开，提示违反github政策，不知道哪里出问题了。后来新注册了一个账号，编译了几次，都没东西出来，只有个配置文件，插件都删光了，最后找了个库，编译了个24兆的文件，太大小米r4g用不了。本地编译可以自己控制，解决问题的手段多一点，感觉比较好。

Henshufu, 2021-11-05:
用本地生成的.config文件,不要连ssh。

WYC-2020, 2021-11-05:
几个月了没失败过

binge8, 2021-11-06:
每天编译2种固件，都成功，两年了没封号

liwenjie119, 2021-11-06:
从来没出过问题，lean官方源码编译一点问题都没有，能正常编译

9l, 2021-11-06:
同时三个action都没事

leopardciaw, 2021-11-06:
> 用本地生成的.config文件,不要连ssh。

SSH确实用不了，一定时间就会停止。想问下，原因是什么哈。

leopardciaw, 2021-11-06:
> 每天编译2种固件，都成功，两年了没封号

编译X86-64的开始还可以，这两天怎么都不行，卡在downloads阶段，不知道什么原因。

alexgzhou, 2021-11-07:
谢谢各位支招，我再试试，nice of you all~

closed 2021-11-07 by alexgzhou

dfldylan, 2022-05-23:
> 用了啊，我的一个主账号开始没注意，同时编译了两个actions，然后被封了，现在还没解开，提示违反github政策，不知道哪里出问题了。后来新注册了一个账号，编译了几次，都没东西出来，只有个配置文件，插件都删光了，最后找了个库，编译了个24兆的文件，太大小米r4g用不了。本地编译可以自己控制，解决问题的手段多一点，感觉比较好。

当然是连接SSH违反了github政策