coolsnowwolf/lede #8191: Newifi3的WIFI問題

DINKHo, 2021-11-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：WIFI设置好可以连上之后，如果重启路由，WIFI不会自动启动，手动去按"重启无线"才能启动连上。
有时WIFI使用一段时间，也会自己停用，也是要手动去按"重启无线"才能启动连上。

### (2) 路由器型号和固件版本

A：Newifi-D2
OpenWrt R21.11.11 / LuCI Master (git-21.280.14233-9b796b1)

### (3) 详细日志

A：
Sat Nov  6 15:20:14 2021 kern.notice kernel: [    0.000000] Linux version 5.4.156 (dink@dink-ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3856-36d254920)) #0 SMP Thu Nov 4 16:05:49 2021
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] MIPS: machine is Newifi-D2
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sat Nov  6 15:20:14 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Nov  6 15:20:14 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Nov  6 15:20:14 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Zone ranges:
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Early memory node ranges
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Sat Nov  6 15:20:14 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
Sat Nov  6 15:20:14 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00031090
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00031090
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] Memory: 509428K/524288K available (6165K kernel code, 252K rwdata, 1292K rodata, 1236K init, 238K bss, 14860K reserved, 0K cma-reserved, 262144K highmem)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Nov  6 15:20:14 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.007827] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.073993] pid_max: default: 32768 minimum: 301
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.078811] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.086012] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Nov  6 15:20:14 2021 kern.info kernel: [    0.096818] rcu: Hierarchical SRCU implementation.
[... 521 lines cut ...]
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.984605] 7
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.986171] 1
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.987734] 1
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.989297] 1
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.990861] 0
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.992460] 1
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.994032] 0
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.995598] 5
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.997161] 9
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  232.998725] 3
Sat Nov  6 15:24:20 2021 kern.warn kernel: [  233.000288] 2
Sat Nov  6 15:24:33 2021 daemon.notice netifd: Network device 'ra0' link is up
Sat Nov  6 15:24:33 2021 daemon.err uhttpd[2202]: ifconfig: SIOCGIFFLAGS: No such device
Sat Nov  6 15:24:33 2021 daemon.err uhttpd[2202]: ifconfig: SIOCGIFFLAGS: No such device
Sat Nov  6 15:24:34 2021 daemon.notice netifd: Network device 'rai0' link is up
Sat Nov  6 15:24:34 2021 daemon.info avahi-daemon[2501]: Joining mDNS multicast group on interface ra0.IPv6 with address fe80::d6ee:7ff:fe61:ac86.
Sat Nov  6 15:24:34 2021 daemon.info avahi-daemon[2501]: New relevant interface ra0.IPv6 for mDNS.
Sat Nov  6 15:24:34 2021 daemon.info avahi-daemon[2501]: Registering new address record for fe80::d6ee:7ff:fe61:ac86 on ra0.*.
Sat Nov  6 15:24:35 2021 daemon.err uhttpd[2202]: brctl: iface rax0: No such device
Sat Nov  6 15:24:35 2021 daemon.info avahi-daemon[2501]: Joining mDNS multicast group on interface rai0.IPv6 with address fe80::d2ee:7ff:fe61:ac86.
Sat Nov  6 15:24:35 2021 daemon.info avahi-daemon[2501]: New relevant interface rai0.IPv6 for mDNS.
Sat Nov  6 15:24:35 2021 daemon.info avahi-daemon[2501]: Registering new address record for fe80::d2ee:7ff:fe61:ac86 on rai0.*.
Sat Nov  6 15:24:37 2021 daemon.info avahi-daemon[2501]: Joining mDNS multicast group on interface apclii0.IPv6 with address fe80::d0ee:7ff:fe60:ac86.
Sat Nov  6 15:24:37 2021 daemon.info avahi-daemon[2501]: New relevant interface apclii0.IPv6 for mDNS.
Sat Nov  6 15:24:37 2021 daemon.info avahi-daemon[2501]: Registering new address record for fe80::d0ee:7ff:fe60:ac86 on apclii0.*.

liwenjie119, 2021-11-06:
这个版本遇到这个问题了，wifi需要网线连上启动，用着用着突然掉线，有时候只是网线连上启动wifi就行，有时候直接死掉，led灯全灭，需要拔电源重启。内核日志只看到下面的，255秒以后的日志多次重复出现，但是没有抓到挂掉前的日志
[    0.000000] Linux version 5.4.156 (runner@fv-az210-419) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3839-d9a8c1657)) #0 SMP Thu Nov 4 16:05:49 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00042186
[    0.000000] Readback ErrCtl register=00042186
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509428K/524288K available (6166K kernel code, 251K rwdata, 1292K rodata, 1236K init, 238K bss, 14860K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 1100MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xfdb1960d81, max_idle_ns: 440795240157 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 3475018995 ns
[    0.000007] sched_clock: 32 bits at 550MHz, resolution 1ns, wraps every 3904515583ns
[    0.007803] Calibrating delay loop... 732.36 BogoMIPS (lpj=3661824)
[    0.073989] pid_max: default: 32768 minimum: 301
[    0.078760] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085956] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096150] rcu: Hierarchical SRCU implementation.
[    0.101405] smp: Bringing up secondary CPUs ...
[    0.121837] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.121847] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.121858] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.121952] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.166178] Synchronize counters for CPU 1: done.
[    2.817141] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.817150] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.817157] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.817204] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.256841] Synchronize counters for CPU 2: done.
[    6.073785] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.073792] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.073801] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.073853] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.342040] Synchronize counters for CPU 3: done.
[    0.371888] smp: Brought up 1 node, 4 CPUs
[    0.379810] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.389596] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.396586] pinctrl core: initialized pinctrl subsystem
[    0.403336] NET: Registered protocol family 16
[... 828 lines cut ...]
[  252.184066] RTMPDrvOpen(2):Check if PDMA is idle!
[  252.188766] WiFi Interface Up Time (rai0): 5.280s
[  252.195199] br-lan: port 3(rai0) entered blocking state
[  252.200556] br-lan: port 3(rai0) entered disabled state
[  252.206455] device rai0 entered promiscuous mode
[  252.211461] br-lan: port 3(rai0) entered blocking state
[  252.216766] br-lan: port 3(rai0) entered forwarding state
[  255.836066] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[  256.049356] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(A4-B1-C1-C7-A1-8C) !!!
[  256.057181] Receive EAPoL-Key frame from STA A4-B1-C1-C7-A1-8C wcid(1)
[... cut ...]
[  272.998382] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(A4-B1-C1-C7-A1-8C) !!!
[  273.006155] Receive EAPoL-Key frame from STA A4-B1-C1-C7-A1-8C wcid(1)
[  273.019437] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(A4-B1-C1-C7-A1-8C) !!!
[  273.027225] Receive EAPoL-Key frame from STA A4-B1-C1-C7-A1-8C wcid(1)
[  273.033867] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(6)=AES
[  273.033867] 
[  273.914749] Rcv Wcid(1) AddBAReq
[  273.918070] Start Seq = 00000017

EnnawYang, 2021-11-07:
表示没遇到过

tuzen2008, 2021-11-11:
> 反馈bug/问题模板，提建议请删除
>
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x]  没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：WIFI设置好可以连上之后，如果重启路由，WIFI不会自动启动，手动去按"重启无线"才能启动连上。 有时WIFI使用一段时间，也会自己停用，也是要手动去按"重启无线"才能启动连上。 [...]

的确有这个问题，重启后WIFI需要手动启动，在第一次连接WIFI无法获取IP，重启WIFI后正常。

VantStark, 2021-11-15:
#8259 
可能是这个问题导致的

DINKHo, 2021-11-15:
> #8259 可能是这个问题导致的

按照这个方法加上ra0 rai0，重启路由2次，WIFI都有自动启动了，看起来是解决问题了！谢谢！

coolsnowwolf, 2026-09-06:
Newifi3 无线问题没有提供无线配置、日志或具体失败现象，且使用旧版本固件，无法在当前 mt76/无线栈中复现，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf