coolsnowwolf/lede #8199: asus RT-ACRH17   opkg 安装ipk运行出错

Remonli, 2021-11-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：github action 编译的固件，去除部分默认选中的luci application后编译，未作其他修改，opkg 安装nano运行出错

### (2) 路由器型号和固件版本

A：asus ACRH17  固件即为 https://github.com/coolsnowwolf/lede 一小时前编译

### (3) 详细日志

A：

```
root@OpenWrt:~# opkg update
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/ipq40xx/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/targets/ipq40xx/generic/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://mirrors.cloud.tencent.com/lede/releases/18.06.9/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg install nano
Installing nano (5.9-1) to root...
Downloading https://mirrors.cloud.tencent.com/lede/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/nano_5.9-1_arm_cortex-a7_neon-vfpv4.ipk
Configuring nano.
root@OpenWrt:~# nano
Error relocating /usr/bin/nano: __stat_time64: symbol not found
Error relocating /usr/bin/nano: __time64: symbol not found
```

Conight, 2021-11-10:
使用编译出来的作为源实施

Remonli, 2021-11-10:
> 
> 
> 使用编译出来的作为源试试

就是编译的默认源啊，没有任何改动的。

Conight, 2021-11-10:
在编译目录下有packages
需要起一个`http server`
然后配置为本地源

Remonli, 2021-11-10:
额，我是GitHub Action 编译的，等我回头装虚拟机本地编译一遍试试看吧。

showme2005, 2021-11-12:
进配置..里面有一个option check_signature在前面加一个# ,保存.在试一下.

closed 2021-11-12 by AmadeusGhost