coolsnowwolf/lede #8200: wan无法连接上游光猫

lon91ong, 2021-11-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 有类似的issue: #7949, 但没有解答

## 2. 详细叙述

### (1) 具体问题

A：路由器通过DHCP获取上游光猫下发的IP, 正常用了一段时间了, 昨天突然无法连接的

### (2) 路由器型号和固件版本

A：极路由B70硬改SPI闪存, 固件版本:5.4.156

### (3) 详细日志

A：系统日志如下:
```
Mon Nov  8 05:00:40 2021 kern.notice kernel: [    0.000000] Linux version 5.4.156 (runner@fv-az50-670) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3856-36d254920)) #0 SMP Wed Nov 3 17:18:18 2021
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] MIPS: machine is HiWiFi HC5962-SPI
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Mon Nov  8 05:00:40 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Mon Nov  8 05:00:40 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mon Nov  8 05:00:40 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Zone ranges:
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000]   HighMem  empty
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Early memory node ranges
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27120 r8192 d22032 u57344
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s27120 r8192 d22032 u57344 alloc=14*4096
Mon Nov  8 05:00:40 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Mon Nov  8 05:00:40 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00025a30
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00025a30
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] Memory: 249956K/262144K available (6278K kernel code, 303K rwdata, 1304K rodata, 1252K init, 249K bss, 12188K reserved, 0K cma-reserved, 0K highmem)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=128 to nr_cpu_ids=4.
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Mon Nov  8 05:00:40 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.007815] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.073986] pid_max: default: 32768 minimum: 301
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.078790] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.085992] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.096742] rcu: Hierarchical SRCU implementation.
Mon Nov  8 05:00:40 2021 kern.info kernel: [    0.102139] smp: Bringing up secondary CPUs ...
[... 416 lines cut ...]
Mon Nov  8 05:45:09 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
...............................
Mon Nov  8 05:45:09 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Mon Nov  8 05:45:09 2021 daemon.notice hostapd: wlan0: ACS-COMPLETED freq=2447 channel=8
Mon Nov  8 05:45:09 2021 kern.info kernel: [ 2690.091140] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mon Nov  8 05:45:09 2021 kern.info kernel: [ 2690.098039] br-lan: port 4(wlan0) entered blocking state
Mon Nov  8 05:45:09 2021 kern.info kernel: [ 2690.103419] br-lan: port 4(wlan0) entered forwarding state
Mon Nov  8 05:45:09 2021 daemon.notice netifd: Network device 'wlan0' link is up
Mon Nov  8 05:45:09 2021 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Mon Nov  8 05:45:09 2021 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Nov  8 05:45:11 2021 daemon.info hostapd: wlan0: STA 0c:a6:94:9f:61:ef IEEE 802.11: authenticated
Mon Nov  8 05:45:11 2021 daemon.info hostapd: wlan0: STA 0c:a6:94:9f:61:ef IEEE 802.11: associated (aid 1)
Mon Nov  8 05:45:11 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 0c:a6:94:9f:61:ef
Mon Nov  8 05:45:11 2021 daemon.info hostapd: wlan0: STA 0c:a6:94:9f:61:ef RADIUS: starting accounting session 81C4323ACC0A1203
Mon Nov  8 05:45:11 2021 daemon.info hostapd: wlan0: STA 0c:a6:94:9f:61:ef WPA: pairwise key handshake completed (RSN)
Mon Nov  8 05:45:11 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 0c:a6:94:9f:61:ef
Mon Nov  8 05:57:10 2021 daemon.info hostapd: wlan0: STA a4:34:a0:92:58:ad IEEE 802.11: authenticated
Mon Nov  8 05:57:10 2021 daemon.info hostapd: wlan0: STA a4:34:a0:92:58:ad IEEE 802.11: associated (aid 3)
Mon Nov  8 05:57:10 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a4:34:a0:92:58:ad
Mon Nov  8 05:57:10 2021 daemon.info hostapd: wlan0: STA a4:34:a0:92:58:ad RADIUS: starting accounting session F515342331EDE89F
Mon Nov  8 05:57:10 2021 daemon.info hostapd: wlan0: STA a4:34:a0:92:58:ad WPA: pairwise key handshake completed (RSN)
Mon Nov  8 05:57:10 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED a4:34:a0:92:58:ad
Mon Nov  8 05:57:11 2021 user.notice nft-qos-monitor: ACTION=add, MACADDR=a4:34:a0:92:58:ad, IPADDR=192.168.77.218, HOSTNAME=M2104K10AC
Mon Nov  8 05:57:11 2021 user.notice nft-qos-dynamic: ACTION=add, MACADDR=a4:34:a0:92:58:ad, IPADDR=192.168.77.218, HOSTNAME=M2104K10AC
```

WYC-2020, 2021-11-08:
先确定光猫是否卡死或者说没法下发ip了

lon91ong, 2021-11-09:
> 先确定光猫是否卡死或者说没法下发ip了

光猫型号是中兴F677, 带有无线, 开无线连手机是正常的

coolsnowwolf, 2026-09-06:
报告基于极路由 B70 硬改 SPI 和旧 5.4.156 固件，WAN DHCP 突然失效没有接口状态、交换机配置或抓包信息，无法定位当前代码问题，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf