coolsnowwolf/lede #8201: PPPOE自动重拨WAN口假死，必须要手动重拨一次才能正常联网。请求大神帮忙看看，谢谢！

mayi5147, 2021-11-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

路由器重启或是运行一段时间后被运营商踢下线，PPPOE重新拨号WAN口会显示已经拨上但是无数据流动，无法联网。必须要重新手动点连接再拨号一次才能正常联网。

### 2.路由器型号和固件版本

路由器是小马v5 X86软路由，安装的ESXI虚拟机，在虚拟机内跑的openwrt。

固件版本是Compiled by Lusty on Oct.12,2021 @ OpenWrt R21.10.1 / LuCI Master (git-21.280.14233-9b796b1)

### 3.附上今日的系统日志，谢谢大神帮忙看看！
Mon Nov  8 03:57:42 2021 daemon.err nlbwmon[12364]: Netlink receive failure: Out of memory
Mon Nov  8 03:57:42 2021 daemon.err nlbwmon[12364]: Unable to dump conntrack: No buffer space available
Mon Nov  8 04:30:49 2021 daemon.err nlbwmon[12364]: Netlink receive failure: Object busy
Mon Nov  8 04:30:49 2021 daemon.err nlbwmon[12364]: Unable to dump conntrack: I/O error
Mon Nov  8 04:36:20 2021 daemon.err nlbwmon[12364]: Netlink receive failure: Object busy
Mon Nov  8 04:36:20 2021 daemon.err nlbwmon[12364]: Unable to dump conntrack: I/O error
Mon Nov  8 04:38:51 2021 daemon.err nlbwmon[12364]: Netlink receive failure: Object busy
Mon Nov  8 04:38:51 2021 daemon.err nlbwmon[12364]: Unable to dump conntrack: I/O error
Mon Nov  8 04:44:52 2021 daemon.err nlbwmon[12364]: Netlink receive failure: Object busy
Mon Nov  8 04:44:52 2021 daemon.err nlbwmon[12364]: Unable to dump conntrack: I/O error
Mon Nov  8 05:04:45 2021 daemon.info pppd[11335]: LCP terminated by peer
Mon Nov  8 05:04:45 2021 daemon.info pppd[11335]: Connect time 2874.9 minutes.
Mon Nov  8 05:04:45 2021 daemon.info pppd[11335]: Sent 1258051323 bytes, received 3189475425 bytes.
Mon Nov  8 05:04:45 2021 daemon.err odhcp6c[11557]: Failed to send RS (Permission denied)
Mon Nov  8 05:04:45 2021 daemon.err odhcp6c[11557]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Mon Nov  8 05:04:45 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Mon Nov  8 05:04:45 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Mon Nov  8 05:04:45 2021 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Mon Nov  8 05:04:45 2021 daemon.notice netifd: wan_6 (11557): Command failed: Permission denied
Mon Nov  8 05:04:45 2021 daemon.notice netifd: Interface 'wan' has lost the connection
Mon Nov  8 05:04:46 2021 daemon.notice pppd[11335]: Modem hangup
Mon Nov  8 05:04:46 2021 daemon.notice pppd[11335]: Connection terminated.
Mon Nov  8 05:04:46 2021 daemon.info pppd[11335]: Connect time 2874.9 minutes.
Mon Nov  8 05:04:46 2021 daemon.info pppd[11335]: Sent 1258051323 bytes, received 3189475425 bytes.
Mon Nov  8 05:04:46 2021 daemon.info pppd[11335]: Sent PADT
Mon Nov  8 05:04:46 2021 daemon.info pppd[11335]: Exit.
Mon Nov  8 05:04:46 2021 daemon.err odhcp6c[11557]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Mon Nov  8 05:04:46 2021 daemon.notice netifd: Interface 'wan_6' is now down
Mon Nov  8 05:04:46 2021 daemon.notice netifd: Interface 'wan_6' is disabled
Mon Nov  8 05:04:46 2021 daemon.notice netifd: Interface 'wan' is now down
Mon Nov  8 05:04:46 2021 daemon.notice netifd: Interface 'wan' is setting up now
Mon Nov  8 05:04:46 2021 daemon.err insmod: module is already loaded - slhc
Mon Nov  8 05:04:46 2021 daemon.err insmod: module is already loaded - ppp_generic
Mon Nov  8 05:04:46 2021 daemon.err insmod: module is already loaded - pppox
Mon Nov  8 05:04:46 2021 daemon.err insmod: module is already loaded - pppoe
Mon Nov  8 05:04:46 2021 daemon.notice netifd: wan (30926): uci: Entry not found
Mon Nov  8 05:04:46 2021 daemon.notice netifd: wan (30926): sh: out of range
Mon Nov  8 05:04:46 2021 daemon.info pppd[30956]: Plugin rp-pppoe.so loaded.
Mon Nov  8 05:04:46 2021 daemon.info pppd[30956]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Mon Nov  8 05:04:46 2021 daemon.notice pppd[30956]: pppd 2.4.8 started by root, uid 0
Mon Nov  8 05:04:49 2021 daemon.info pppd[30956]: PPP session is 17894
Mon Nov  8 05:04:49 2021 daemon.warn pppd[30956]: Connected to c4:b8:b4:01:bb:7a via interface eth2
Mon Nov  8 05:04:49 2021 kern.info kernel: [685584.200752] pppoe-wan: renamed from ppp0
Mon Nov  8 05:04:49 2021 daemon.info pppd[30956]: Renamed interface ppp0 to pppoe-wan
Mon Nov  8 05:04:49 2021 daemon.info pppd[30956]: Using interface pppoe-wan
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: Connect: pppoe-wan <--> eth2
Mon Nov  8 05:04:49 2021 daemon.info pppd[30956]: syncppp not active
Mon Nov  8 05:04:49 2021 daemon.info pppd[30956]: CHAP authentication succeeded: Authentication success,Welcome!
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: CHAP authentication succeeded
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: peer from calling number C4:B8:B4:01:BB:7A authorized
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: local  LL address fe80::ccf2:0c6f:0e3a:d4a0
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: remote LL address fe80::c6b8:b4ff:fe01:bb7a
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: local  IP address 100.64.201.17
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: remote IP address 100.64.0.1
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: primary   DNS address 211.137.58.20
Mon Nov  8 05:04:49 2021 daemon.notice pppd[30956]: secondary DNS address 211.137.64.163
Mon Nov  8 05:04:49 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
[... 1207 lines cut ...]
[   12.288721] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   21.652656] br-lan: port 1(eth0) entered disabled state
[   21.657342] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   21.658706] br-lan: port 1(eth0) entered blocking state
[   21.658852] br-lan: port 1(eth0) entered forwarding state
[   21.745605] e1000: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   21.825356] e1000: eth2 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   21.935443] e1000: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   22.740508] pppoe-wan: renamed from ppp0
[  219.200121] pppoe-wan: renamed from ppp0
[253541.647209] pppoe-wan: renamed from ppp0
[253842.754375] pppoe-wan: renamed from ppp0
[340003.496931] pppoe-wan: renamed from ppp0
[340355.903770] pppoe-wan: renamed from ppp0
[426375.695890] pppoe-wan: renamed from ppp0
[426678.253342] pppoe-wan: renamed from ppp0
[512756.575877] pppoe-wan: renamed from ppp0
[513072.188279] pppoe-wan: renamed from ppp0
[685584.200752] pppoe-wan: renamed from ppp0
[685856.759643] pppoe-wan: renamed from ppp0
[697921.682063] ssr-redir[29461]: segfault at 158 ip 00000000004059b5 sp 00007ffdcfd50d50 error 4 in ssr-redir[403000+14000]
[697921.682412] Code: 72 0e 48 39 c6 73 0b 48 8d 0c 16 48 39 c8 73 02 0f 0b 48 89 c7 48 89 d1 f3 a4 c3 48 89 f7 ff 25 a9 a5 01 00 55 53 48 89 fb 52 <48> 8b af 58 01 00 00 48 8b 7d 08 48 85 ff 74 06 ff 15 8d a5 01 00
[697921.702401] ssr-redir[29463]: segfault at 158 ip 00000000004059b5 sp 00007ffffc414330 error 4 in ssr-redir[403000+14000]
[697921.703025] Code: 72 0e 48 39 c6 73 0b 48 8d 0c 16 48 39 c8 73 02 0f 0b 48 89 c7 48 89 d1 f3 a4 c3 48 89 f7 ff 25 a9 a5 01 00 55 53 48 89 fb 52 <48> 8b af 58 01 00 00 48 8b 7d 08 48 85 ff 74 06 ff 15 8d a5 01 00
[697986.820853] pppoe-wan: renamed from ppp0

WYC-2020, 2021-11-08:
目前我没遇到过，我也是拨号上网的

mayi5147, 2021-11-08:
测试了一天发现刚刚启动后是可以正常拨上去的，然后大概5到10秒后就返回超时了。

Suwipe, 2021-11-09:
这个问题两个多月了，电信每2天强制短线自动重拨就会出现wan_6丢失获取不到IP，手动重启就从没遇到过，自己写了个hotplug脚本检查解决，但是最近又出现更大的问题，运营商强制短线自动重拨后lan口也出问题，DHCP崩了，设备获取不到v4地址，同时dnsforwarder也会崩，自动退出，重启LAN口解决DHCP问题，好想回到8月的版本，那时候的基本没什么问题

laaArmy, 2021-11-11:
我把wan6删除后问题解决了
删除后对IPV6没有影响

mayi5147, 2021-11-12:
我现在也是删除了wan6然后禁用了ipv6问题得到解决，重启不会再连接wan_6这个接口了。

zzmo, 2021-11-13:
R21.11.11刷完固件第一次正常。pppoe重播之后可以获取ip，但上不了网，lan口dhcp不能自动续租。没有wan6 也没有ipv6。上一次九月份跑出来的没有这个情况。

jxph1123, 2021-11-30:
> 这个问题两个多月了，电信每2天强制短线自动重拨会出现wan_6丢失获取不到IP，手动重启就从没遇到过，自己写了个热插拔脚本检查，但最近又出现了更新检查大的问题，运营商强制短线自动重拨后lan口也出问题，DHCP崩了，设备未获取v4地址，同时dnsforwarder也会崩塌，自动退出，重启LAN口解决DHCP问题，好想回到8月的版本，思想的基本问题
我最近也出现这个问题，同样的版本，以前可以一个月不重启，最近频繁断网。。我怀疑供应商的毛病，我移动网。

Jerrycelery, 2021-12-31:
> 

我也是移动的网络,最近频繁断网,现象为家里的设备无法自动获取IP, 隔几天就会断网,重启软路由后解决.非常烦人这个问题.不知道你解决没?我的IPV6在OP你禁用了,WAN口那边都不显示IPV6的拨号信息了,应该跟这个没关系吧

82kg, 2022-11-12:
ssh或TTYD进去
在/etc/ppp/options中加入下面两行
`maxfail 0`
`persist`

closed 2026-09-05 by coolsnowwolf