coolsnowwolf/lede #8209: 云编译qBittorrent-static下载失败

stock169, 2021-11-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：x86的可以下载编译成功，竞斗云arm的不行下载失败退回上一版就可以下载编译成功，都是云编译不存在科学问题的

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志
make download 里的
A： ERROR: package/lean/qBittorrent-static failed to build.
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

后面编译部分错误日志也不贴了就是尝试了所有镜像链接都是404失败最后报错退出

HanzChen, 2021-11-09:
network里去掉qbit的选项就好了

ZeaKyX, 2021-11-10:
没有看到你的云编译项目，我想看一下错误的链接
release里armv7的包是有的，核对一下链接是不是https://github.com/userdocs/qbittorrent-nox-static/releases/download/release-4.3.9_v1.2.14/armv7-qbittorrent-nox
[image]

stock169, 2021-11-10:
> https://github.com/userdocs/qbittorrent-nox-static/releases/download/release-4.3.9_v1.2.14/armv7-qbittorrent-nox [image]

在后面的编译log里我记得很清楚下载链接里没有armv7-开头，我估计也是这个问题但奇怪的是我sed -i 's/4.3.9_v1.2.14/4.3.8_v1.2.14/g' package/lean/qBittorrent-static/Makefile 改一个版本号就能通过这实在不科学

ZeaKyX, 2021-11-10:
不好意思我应该搞错了，竞斗云应该是`armhf`，源确实没有没再提供了
[image]

closed 2021-11-12 by AmadeusGhost