coolsnowwolf/lede #8210: Ubuntu20.04lts初次编译k2p固件失败

github-logic, 2021-11-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [√] 没有类似的issue
仅使用lede源码，根据github的教程，每次编译错误都是这个，编译所用的机器是国外的一家vps，系统是Ubuntu2 0.04 LTS

g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -static-libstdc++ -static-libgcc  -no-pie -o build/genautomata \
    build/genautomata.o build/rtl.o build/read-rtl.o build/ggc-none.o build/vec.o build/min-insn-modes.o build/gensupport.o build/print-rtl.o build/hash-table.o build/read-md.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a -lm
build/genautomata /home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/common.md /home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/config/mips/mips.md \
  insn-conditions.md > tmp-automata.c
/bin/bash: line 1: 785366 Killed                  build/genautomata /home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/common.md /home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0/gcc/config/mips/mips.md insn-conditions.md > tmp-automata.c
make[5]: *** [Makefile:2345: s-automata] Error 137
make[5]: Leaving directory '/home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/gcc'
make[4]: *** [Makefile:4251: all-gcc] Error 2
make[4]: Leaving directory '/home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial'
make[3]: *** [Makefile:37: /home/logic/lede/build_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/gcc-8.4.0-initial/.built] Error 2
make[3]: Leaving directory '/home/logic/lede/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#256.88#30.76#297.21
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/logic/lede'
make[1]: *** [toolchain/Makefile:93: /home/logic/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/logic/lede'
make: *** [/home/logic/lede/include/toplevel.mk:230: world] Error 2

AmadeusGhost, 2021-11-09:
你机子内存不够，单线程编译吧

deplives, 2021-11-09:
[image]

Cathgao, 2021-11-09:
不会是512MB内存的机器吧，编译那个node，起码都要1GB，至少得有SWAP

closed 2021-11-12 by AmadeusGhost