coolsnowwolf/lede #8220: MT7615 DBDC 编译冲突问题

JKSTAFF, 2021-11-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在单7615DBDC设备上编译Kmod和mtwifi的luci时，会与wireless-tool产生依赖冲突。具体情况详见（3）

### (2) 路由器型号和固件版本

A：Glinet MT300，该错误在openwrt官方和lean仓库master分支编译固件时都会发生

### (3) 详细日志

A：在包含了7615D和DBDC并同时使用mtwifi的luci时，wireless-tools会变成{M}选中状态，随即编译会产生如下错误：  
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency wireless-tools for mt_wifi
 * pkg_hash_fetch_best_installation_candidate: Packages for mt_wifi found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mt_wifi.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-mtwifi:
 *      wireless-tools
 * opkg_install_cmd: Cannot install package luci-app-mtwifi
```
此时若强制选中 wireless-tools 呈现为{*}选中标识，编译会产生如下错误：  
```
Collected errors:
 * check_conflicts_for: The following packages conflict with mt_wifi:
 * check_conflicts_for:         kmod-mt7615d_dbdc * 
 * opkg_install_cmd: Cannot install package mt_wifi.
 * check_conflicts_for: The following packages conflict with mt_wifi:
 * check_conflicts_for:         kmod-mt7615d_dbdc * 
 * opkg_install_cmd: Cannot install package luci-app-mtwifi.
```

AmadeusGhost, 2021-11-10:
选 kmod-mt7615d_dbdc 就好了，其它包不要选。

JKSTAFF, 2021-11-10:
> 选 kmod-mt7615d_dbdc 就好了，其它包不要选。
@AmadeusGhost 更多细节确认希望🤝。当我在内核模块中选中kmod-mt7615d_dbdc时，kmod-mt7615d是会被自动选中的。  
Nossiac那边对闭源驱动的描述是不兼容原来的luci界面和hostapd这些的，所以L大这边也准备了luci-app-mtwifi这样的图形界面，请问这个包不能选择么？  
紧接着上面的问题，如果我需要使用luci-app-mtwifi作为闭源驱动的图形界面，那么当我选中这个包的时候，basesystem下的wireless-tool会被自动选为{M}并且不可取消选择。因此我不太明白您说的其他包不要选所指。  
急切的期待您的解答

AmadeusGhost, 2021-11-10:
不需要选 mtwifi，选那个内核模块就好了。

JKSTAFF, 2021-11-10:
> 不需要选 mtwifi，选那个内核模块就好了。

@AmadeusGhost 目前是没有图形界面支持了是么

AmadeusGhost, 2021-11-10:
不是，你试试就知道了。。

JKSTAFF, 2021-11-10:
感谢指导。luci原生界面可用，无线已正常工作

nanchuci, 2021-11-11:
参考K2P和jcg y2来配置-luci-newapi -wpad-openssl kmod-mt7615d_dbdc，当然还要修改改驱动添加进你的机型！
这样做的好处是用大雕源码的开源无线界面来配置无线，坏处就是5G信道要2G信道设置为手动后才会生效，还有大雕的开源LUCI好久没更新了，虽然@jjm2473后来修改了对闭源驱动的支持，但是管理界面还是不是很完善！

closed 2021-11-12 by AmadeusGhost

jjm2473, 2021-11-13:
mt7615d_dbdc 支持用官方的luci配置，不需要再安装 luci-app-mtwifi

jjm2473, 2021-11-13:
@JKSTAFF 请参考这个PR https://github.com/coolsnowwolf/lede/pull/5844

Hess488, 2022-05-11:
> mt7615d_dbdc 支持用官方的luci配置，不需要再安装 luci-app-mtwifi

K2p驱动默认勾选的是7615d_dbdc跟7615d。这是闭源驱动吗，这种情况下是不是不需要勾选luci-app-mtwifi也是闭源驱动了

jjm2473, 2022-05-11:
只要使用7615d_dbdc驱动都可以直接使用官方的luci配置，不需要再使用luci-app-mtwifi，而且官方luci功能还更丰富。具体参考k2p相关的改动

Hess488, 2022-05-11:
> 只要使用7615d_dbdc驱动都可以直接使用官方的luci配置，不需要再使用luci-app-mtwifi，而且官方luci功能还更丰富。具体参考k2p相关的改动

还有两个问题不懂希望大佬解答一下万分感谢

k2p编译时在默认不改动任何驱动配置，也不勾选luci-app-mtwifi的情况下编译出来的固件是闭源驱动还是开源驱动

我编译k2p想使用闭源驱动需要怎么修改，关闭哪些开启哪些选项

jjm2473, 2022-05-12:
https://github.com/coolsnowwolf/lede/blob/0b8b34215fd937a70f4d056ff575708bd3207e9d/target/linux/ramips/image/mt7621.mk#L1242-L1250
@Hess488 k2p目前默认就是闭源驱动，应该也没选中luci-app-mtwifi。

Hess488, 2022-05-12:
> https://github.com/coolsnowwolf/lede/blob/0b8b34215fd937a70f4d056ff575708bd3207e9d/target/linux/ramips/image/mt7621.mk#L1242-L1250
> 
> @Hess488 k2p目前默认就是闭源驱动，应该也没选中luci-app-mtwifi。

好的谢谢，k2p默认就是闭源驱动编译。那也就是不需要选luci-app-mtwifi这个插件直接编译就行了是吧

jjm2473, 2022-05-12:
嗯

msdos03, 2022-10-15:
@jjm2473 除了k2p以外的机型可以用7615d_dbdc驱动并且用luci直接配置吗？我的是msg1500，无线部分是单个mt7615dn。

jjm2473, 2022-10-15:
> 

> @jjm2473 除了k2p以外的机型可以用7615d_dbdc驱动并且用luci直接配置吗？我的是msg1500，无线部分是单个mt7615dn。

不了解，我的路由器只有k2p才用dbdc驱动，当时只是为了适配自己的路由器才改的。我有其他mt7621方案的路由器，例如newifi 2 和 newifi3，都不是使用dbdc驱动的。
其他设备的问题还是问这个仓库专业的开发者吧，例如lean，天灵等