coolsnowwolf/lede #8221: mt76x2u usb网卡

mingcz0301, 2021-11-10:
## 1.关于你要提交的问题
R4S 4G  +  USB网卡 A6210 MT76x2U  WIFI 问题

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

## 2. 详细叙述
R4S 配合 网件A6210 网卡做AP,
每次重启后路由器wifi 可以正常使用, 几分钟后 wifi 断线,重连提示密码错误,或无法加入

之前 5.10.78+2021-10-23.1-..a-4 是可以正常使用的

### (2) 路由器型号和固件版本

A：r4s   2021-11-10 clone 

### (3) 详细日志

A：
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.158 (mingcz@iZb5cc5coco2z0Z) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3870-04d5160fa)) #0 SMP PREEMPT Tue Nov 9 00:06:48 2021
[    0.000000] Machine model: FriendlyElec NanoPi R4S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x00000000f7800000
[    0.000000] On node 0 totalpages: 1015296
[    0.000000]   DMA32 zone: 15864 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 1015296 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 20 pages/cpu s44888 r8192 d28840 u81920
[    0.000000] pcpu-alloc: s44888 r8192 d28840 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff1a0000 root=PARTUUID=d6f1695b-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3959760K/4061184K available (7422K kernel code, 540K rwdata, 2332K rodata, 1664K init, 716K bss, 93232K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: no VLPI support, no direct LPI support
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @f2080000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x00000000f2040000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x00000000f2050000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x39c with crng_init=0
[... 427 lines cut ...]
[  145.190017] x15: 0000000000000000 x14: fdc187e3cd5da1ac 
[  145.190479] x13: ffff000000000000 x12: ffff0000e79d9000 
[  145.190942] x11: 071c71c71c71c71c x10: 0000000000000880 
[  145.191404] x9 : ffff800011793d90 x8 : ffff0000edc166e0 
[  145.191867] x7 : 0000000000000001 x6 : 00000000d9ba587b 
[  145.192329] x5 : 00ffffffffffffff x4 : 0000000000000000 
[  145.192792] x3 : 0000000000000001 x2 : ffff0000edc15e00 
[  145.193254] x1 : 0000000000000000 x0 : 0000000000000001 
[  145.193717] Call trace:
[  145.193934]  __mt76_worker_fn+0xf0/0x1e0 [mt76]
[  145.194335]  kthread+0x14c/0x150
[  145.194618]  ret_from_fork+0x10/0x1c
[  145.194932] Code: 17ffffe6 1400000e 17ffffde f9800271 (c85f7e60) 
[  145.195464] ---[ end trace 339ba1b38c5e6950 ]---
[  195.180054] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
[  214.247275] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[  214.248022] br-lan: port 1(eth1) entered blocking state
[  214.248495] br-lan: port 1(eth1) entered forwarding state
[  242.190102] br-lan: port 1(eth1) entered disabled state
[  242.192774] r8169 0000:01:00.0 eth1: Link is Down
[  248.359603] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[  278.575254] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
[  295.190491] r8169 0000:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[  295.191195] br-lan: port 1(eth1) entered blocking state
[  295.191667] br-lan: port 1(eth1) entered forwarding state

mingcz0301, 2021-11-18:
有没有大佬能帮忙分析下这个问题..

dlliuyisheng, 2021-12-23:
我出现的问题是无法使用A6210进入AP模式，请问这是什么问题？

SuperChen-CC, 2023-02-08:
同 R4S + 网件A6210

尝试后发现只有如下配置可以开启

| 模式 | 带宽 | 信道 | 频率 |
| -- | -- | -- | -- |
| N | 5GHz | <64 | 40MHz |

速度大概能跑的 20MB/s，基本能满足需求。

2457117265, 2023-07-10:
> 同 R4S + 网件A6210
> 
> 尝试后发现如下配置可以开启5G
> 
> 模式	带宽	海峡	频率
> 氮	5GHz	36	40兆赫
> 速度大概能跑到20MB/s，基本能满足需求。

可以分享下固件吗

closed 2026-09-05 by coolsnowwolf