coolsnowwolf/lede #8229: HiWiFi HC5962 有时候自动重启的问题

hzjixiang, 2021-11-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有此类问题，多发生于MT7621芯片
## 2. 详细叙述

### (1) 具体问题

A：莫名奇妙重启，频率两三天一次

### (2) 路由器型号和固件版本

A：HC5962 内核5.4.150 R21.10.1 

### (3) 详细日志

A：Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.865271] [    723]     0   723      236        6    16384        0             0 askfirst
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.873788] [   1939]   514  1939      317       32    20480        0             0 logd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.881953] [   1940]     0  1940      351       21    16384        0             0 logread
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.890361] [   1988]     0  1988      403       22    16384        0             0 rpcd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.898533] [   2358]     0  2358     1329       73    24576        0             0 hostapd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.906960] [   2359]     0  2359     1295       57    20480        0             0 wpa_supplicant
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.916058] [   2675]     0  2675      390       41    16384        0             0 odhcpd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.924417] [   4546]     0  4546      300       14    12288        0             0 vsftpd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.932779] [   4603]     0  4603      375       20    20480        0             0 dbus-daemon
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.941552] [   4721]     0  4721      609       63    16384        0             0 smbd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.949700] [   4722]     0  4722      626       15    20480        0             0 nmbd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.957865] [   4835] 65534  4835      510       12    16384        0             0 avahi-daemon
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.966724] [   4909]     0  4909      242        8    16384        0             0 vlmcsd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.975063] [   6380]     0  6380      487      140    24576        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.983055] [   7101]     0  7101      314       11    16384        0             0 ntpd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.991220] [   7571]     0  7571      450       39    20480        0             0 netifd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826099.999541] [   9987]     0  9987      458        9    16384        0             0 starter
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.007986] [  10127]     0 10127     1465       70    20480        0             0 charon
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.016326] [  10416]     0 10416      288       12    12288        0             0 dropbear
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.024902] [  11817]     0 11817     1264       67    24576        0             0 ttyd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.033087] [  13045]     0 13045      314        9    16384        0             0 crond
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.041343] [  20273]     0 20273      268        7    12288        0             0 odhcp6c
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.049749] [  20319]     0 20319      325       26    16384        0             0 pppd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.057913] [  20533]     0 20533      268       12    16384        0             0 odhcp6c
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.066338] [  21766]     0 21766      631        0    20480        0             0 nlbwmon
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.074786] [  26218]     0 26218     1011       83    16384        0             0 ssr-redir
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.083387] [  26220]     0 26220     1016       87    20480        0             0 ssr-redir
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.091985] [  26222]     0 26222     1011       81    24576        0             0 ssr-redir
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.100566] [  26224]     0 26224     1006       77    24576        0             0 ssr-redir
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.109162] [  26458] 65534 26458      472       93    20480        0             0 pdnsd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.117428] [  26661]     0 26661      340       36    16384        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.125492] [  29953]     0 29953      325       22    20480        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.133503] [  29972]     0 29972      364       62    20480        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.141522] [  29976]     0 29976      313        9    16384        0             0 lock
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.149672] [  29990]     0 29990      364       61    20480        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.157664] [  29991]     0 29991      527       11    12288        0             0 ip
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.165658] [  29992]     0 29992      313        7    16384        0             0 awk
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.173739] [  29993]     0 29993      313        7    16384        0             0 egrep
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.181989] [  30012]     0 30012      313        7    16384        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.189963] [  30014]     0 30014      313        9    16384        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.198015] [  30015]     0 30015      313        7    12288        0             0 ifconfig
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.206539] [  30016]     0 30016      313        7    16384        0             0 grep
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.214723] [  30018]     0 30018      313        7    12288        0             0 awk
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.222842] [   6112]     0  6112      313        7    16384        0             0 sleep
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.231173] [   6253]     0  6253     1168      851    24576        0             0 dnsmasq
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.239611] [   6261]     0  6261     1087       54    24576        0             0 uhttpd
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.247960] [   6264]     0  6264      277       17    16384        0             0 modprobe
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.256476] [   6265]     0  6265      340       36    16384        0             0 sh
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.264469] [   6266]     0  6266      313        7    12288        0             0 ps
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.272465] [   6267]     0  6267      277        1    12288        0             0 grep
Thu Nov 11 03:15:33 2021 kern.info kernel: [826100.280612] [   6268]     0  6268      277        1    16384        0             0 grep
[... 198 lines cut ...]
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.483064] [  20273]     0 20273      268        7    12288        0             0 odhcp6c
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.491491] [  20319]     0 20319      325       26    16384        0             0 pppd
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.499631] [  20533]     0 20533      268       12    16384        0             0 odhcp6c
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.508073] [  21766]     0 21766      631        0    20480        0             0 nlbwmon
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.516498] [  26218]     0 26218     1011       83    16384        0             0 ssr-redir
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.525096] [  26220]     0 26220     1016       87    20480        0             0 ssr-redir
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.533695] [  26222]     0 26222     1011       81    24576        0             0 ssr-redir
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.542297] [  26224]     0 26224     1006       77    24576        0             0 ssr-redir
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.550896] [  26458] 65534 26458      472       93    20480        0             0 pdnsd
[... cut ...]
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.648318] [  30016]     0 30016      313        7    16384        0             0 grep
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.656484] [  30018]     0 30018      313        7    12288        0             0 awk
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.664599] [   6112]     0  6112      313        7    16384        0             0 sleep
Thu Nov 11 03:16:11 2021 kern.info kernel: [826136.672862] [   6253]     0  6253     1168        0    24576        0             0 dnsmasq
之后重启

AmadeusGhost, 2021-11-16:
东西开太多 oom 了？

hzjixiang, 2021-11-16:
> 东西开太多 oom 了？

不多。ssr+ssr_mudb_server+DDNS+后台页面端口转发。  RAM空闲160M

blueveryday, 2021-11-20:
我也是这个机型，没有遇到你的问题。
不过最新的11.11代码pppoe无法拨号上网，换回以前的版本了。

ddd-zero, 2022-02-16:
你开 ACC里面的HWNAT了吗？
#8791

coolsnowwolf, 2026-09-06:
HC5962 每数日重启只有进程列表，没有 watchdog、内核 oops、内存和电源证据，无法定位，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf