coolsnowwolf/lede #8231: 编译加上mariadb出错

hy021, 2021-11-11:
无论是直接拉取的mariadb10.4.18-1和换了10.4.13都过不了错误信息如下：
-- MariaDB 10.4.13
CMake Deprecation Warning at wsrep-lib/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Wsrep-lib version: 1.0.0
== Configuring MariaDB Connector/C
CMake Deprecation Warning at libmariadb/CMakeLists.txt:5 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR) 
-- Dynamic column API support: ON
CMake Error at libmariadb/cmake/ConnectorName.cmake:30 (ENDMACRO):
  Flow control statements are not properly nested.                                                                            
Call Stack (most recent call first):                                                                                          
  libmariadb/CMakeLists.txt:434 (INCLUDE)                                                                                     
                                                                                                                              
                                                                                                                              
-- Configuring incomplete, errors occurred!
See also "/home/nacy/lede/build_dir/hostpkg/mariadb-10.4.13/CMakeFiles/CMakeOutput.log".
See also "/home/nacy/lede/build_dir/hostpkg/mariadb-10.4.13/CMakeFiles/CMakeError.log".
make[3]: *** [Makefile:594: /home/nacy/lede/build_dir/hostpkg/mariadb-10.4.13/.configured] Error 1
make[3]: Leaving directory '/home/nacy/lede/feeds/packages/utils/mariadb'
time: package/feeds/packages/mariadb/host-compile#0.59#1.76#2.53
    ERROR: package/feeds/packages/mariadb [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/mariadb/host/compile] Error 1
make[2]: Leaving directory '/home/nacy/lede'
make[1]: *** [package/Makefile:110: /home/nacy/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nacy/lede'
make: *** [/home/nacy/lede/include/toplevel.mk:230：world] 错误 2

切到R21.6.1分支编译MariaDB 10.4.13是没问题的。

Boos4721, 2021-11-11:
[image]
**WORKS**

hy021, 2021-11-11:
> [image] **WORKS**
mariadb10.4.18编译过了，但是无法启动，cd进入/etc/init.d直接mysqld不加start可以启动，但会出现假死，其实已经启动了，这样就导至守护进程成了摆设，没法开机自启。

closed 2022-02-12 by Beginner-Go