coolsnowwolf/lede #8233: 缺文件，编译失败

ostrichii, 2021-11-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
更新代码后变失败
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
R4S,5.10.78内核
A：

### (3) 详细日志

A：
make[5]: Leaving directory '/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78'
cmp -s /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/vmlinux /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.debug || { aarch64-openwrt-linux-musl-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/vmlinux /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux; aarch64-openwrt-linux-musl-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/vmlinux /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.elf; cp -fpR /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/vmlinux /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/vmlinux.debug;  cp -fpR /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot//Image /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/Image;  }
touch /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/ostrich/r4s/lede/target/linux/rockchip/image'
cp -fpR /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/COPYING /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/COPYING.linux
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/include -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/include/ -undef -D__DTS__  -o /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/scripts/dtc/dtc -O dtb -i/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:417.3-13: Warning (reg_format): /usb@ff600000/dwc3@ff600000/usb-eth@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:415.12-420.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #address-cells value
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:415.12-420.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #size-cells value
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1067.25-1078.5: Warning (interrupt_provider): /pinctrl/gpio0@ff210000: Missing #address-cells in interrupt provider
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1080.25-1091.5: Warning (interrupt_provider): /pinctrl/gpio1@ff220000: Missing #address-cells in interrupt provider
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1093.25-1104.5: Warning (interrupt_provider): /pinctrl/gpio2@ff230000: Missing #address-cells in interrupt provider
/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3328.dtsi:1106.25-1117.5: Warning (interrupt_provider): /pinctrl/gpio3@ff240000: Missing #address-cells in interrupt provider
rm -f /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/include -I/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/include/ -undef -D__DTS__  -o /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts
aarch64-openwrt-linux-musl-cpp: error: /home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
Makefile:78: recipe for target '/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb' failed
make[5]: *** [/home/ostrich/r4s/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb] Error 1
make[5]: Leaving directory '/home/ostrich/r4s/lede/target/linux/rockchip/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/ostrich/r4s/lede/target/linux/rockchip'
Makefile:11: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/ostrich/r4s/lede/target/linux'
time: target/linux/install#6.49#1.81#8.35
    ERROR: target/linux failed to build.
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 1
make[2]: Leaving directory '/home/ostrich/r4s/lede'
target/Makefile:19: recipe for target '/home/ostrich/r4s/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
make[1]: *** [/home/ostrich/r4s/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ostrich/r4s/lede'
/home/ostrich/r4s/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

KLT007, 2021-11-11:
删除build_dir重新编译

ostrichii, 2021-11-11:
> 删除build_dir重新编译

不是这个问题

KLT007, 2021-11-11:
选doornet2在5.10编译就缺失。

GreatMichaelLee, 2021-11-11:
Same issue, how to resolve? 

rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/include -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/include/ -undef -D__DTS__  -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts
aarch64-openwrt-linux-musl-cpp: error: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.78/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.

KLT007, 2021-11-11:
> Same issue, how to resolve?
>
> rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp aarch64-openwrt-linux-musl-cpp -nostdinc -x [...]

5.10 kernel does not have doornet2, why compile it?

ostrichii, 2021-11-11:
> 选doornet2在5.10编译就缺失。

一直编译的5.10，没刻意选这玩意儿，也没改配置，拉取新代码之后就出现了这个问题，也没看到在哪里取消勾选。

GreatMichaelLee, 2021-11-11:
> > 选doornet2在5.10编译就缺失。
> 
> 一直编译的5.10，没刻意选这玩意儿，也没改配置，拉取新代码之后就出现了这个问题，也没看到在哪里取消勾选。

Yes, I once remember in last month or some time before, there occured the similar problem(doornet2 files missing), and later on the developer fix it by adding the missing files

KLT007, 2021-11-11:
> > 选doornet2在5.10编译就缺失。
> 
> 一直编译的5.10，没刻意选这玩意儿，也没改配置，拉取新代码之后就出现了这个问题，也没看到在哪里取消勾选。

可以解决方法就是去target/linux/rockchip/patches-5.4把https://github.com/coolsnowwolf/lede/blob/master/target/linux/rockchip/patches-5.4/205-rockchip-rk3399-Add-support-for-EmbedFire-DoorNet2.patch 复制一份到patches-5.10目录下，应该可以

KLT007, 2021-11-11:
在lede目录下执行 wget -P target/linux/rockchip/patches-5.10/ https://raw.githubusercontent.com/coolsnowwolf/lede/master/target/linux/rockchip/patches-5.4/205-rockchip-rk3399-Add-support-for-EmbedFire-DoorNet2.patch  应该可以编译成功吧，没试过

ostrichii, 2021-11-11:
> 在lede目录下执行 `wget -P target/linux/rockchip/patches-5.10/ https://raw.githubusercontent.com/coolsnowwolf/lede/master/target/linux/rockchip/patches-5.4/205-rockchip-rk3399-Add-support-for-EmbedFire-DoorNet2.patch` 应该可以编译成功吧，没试过

我试试

ostrichii, 2021-11-11:
> 在lede目录下执行 `wget -P target/linux/rockchip/patches-5.10/ https://raw.githubusercontent.com/coolsnowwolf/lede/master/target/linux/rockchip/patches-5.4/205-rockchip-rk3399-Add-support-for-EmbedFire-DoorNet2.patch` 应该可以编译成功吧，没试过

这样可以了

KLT007, 2021-11-11:
> > 在lede目录下执行 `wget -P target/linux/rockchip/patches-5.10/ https://raw.githubusercontent.com/coolsnowwolf/lede/master/target/linux/rockchip/patches-5.4/205-rockchip-rk3399-Add-support-for-EmbedFire-DoorNet2.patch` 应该可以编译成功吧，没试过
> 
> 这样可以了，辛苦开发者们看到合一下代码，谢谢

https://github.com/coolsnowwolf/lede/pull/8232/commits/d29472d759a8458fb5db16f864be6502010455fa

LPDDR6-10000MHz, 2021-11-11:
已合并了代码，记得关闭评论😘

ostrichii, 2021-11-11:
> 已合并了代码，记得关闭评论😘

好的大佬

closed 2021-11-11 by ostrichii