coolsnowwolf/lede #8235: unblockmusic 失效了？

doumeki, 2021-11-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
NO
## 2. 详细叙述

root@x~: **lsof -iTCP:5200 -iTCP:5201**
COMMAND     PID USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
UnblockNe 14562 root   10u  IPv6 2856446      0t0  TCP *:5200 (LISTEN)
UnblockNe 14562 root   12u  IPv6 2857137      0t0  TCP *:5201 (LISTEN)
root@x~: **pgrep -laf UnblockNe**
14562 UnblockNeteaseMusic -o kuwo:kugou -b -p 5200 -sp 5201 -c /usr/share/UnblockNeteaseMusicGo/server.crt -k /usr/share/UnblockNeteaseMusicGo/server.key -m 0 -e -sl 0 -l /tmp/unblockmusic.log
14809 /bin/sh /usr/share/UnblockNeteaseMusic/logcheck.sh
root@DMKLIGHT ~:

**日志：没有任何音乐信息**
Thu, 11 Nov 2021 19:41:31 +0800 # Start UnblockNeteaseMusic
Thu, 11 Nov 2021 19:41:31 +0800 # UnblockNeteaseMusic Golang Version (http:5200, https:5201)
--
	
全灰的
[image]

Beginner-Go, 2021-11-11:
用node版本

doumeki, 2021-11-11:
> 用node版本

试了node版本也不行,最后要在云音乐里强制使用代理才行,我记得之前是不用的.

doumeki, 2021-11-11:
破案了，用了最新的beta 版本的adguardhome, 某个规则把163给当成AD了， 放行就可以了
[image]

closed 2021-11-11 by doumeki