coolsnowwolf/lede #8239: 数据类型溢出

ostrichii, 2021-11-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
网络-接口状态信息 数据包显示负数，之前提了x86大雕修过
### (1) 具体问题

A：
[image]

### (2) 路由器型号和固件版本
R4S
OpenWrt R21.11.11
A：

### (3) 详细日志

A：

WYC-2020, 2021-11-30:
[image]
从这里开始做手脚吧

ostrichii, 2021-12-01:
> [image] 从这里开始做手脚吧

多谢大佬，我改改看

ostrichii, 2021-12-01:
哪位大佬看到帮fix一下
%d -> %f

feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_network/iface_overview.htm
144 '<strong><%:RX%></strong>: %.2mB (%d <%:Pkts.%>)<br />' +
145 '<strong><%:TX%></strong>: %.2mB (%d <%:Pkts.%>)<br />'

feeds/luci/modules/luci-mod-admin-full/luasrc/view/admin_network/iface_status.htm
 34 '<strong><%:RX%></strong>: %.2mB (%d <%:Pkts.%>)<br />' +
 35 '<strong><%:TX%></strong>: %.2mB (%d <%:Pkts.%>)<br />',

WYC-2020, 2021-12-02:
改成%s不是兼容更好 @ostrichii

ostrichii, 2021-12-02:
> 改成%s不是兼容更好 @ostrichii

我觉得大佬说得对

coolsnowwolf, 2026-09-06:
该报告针对 R21.11 的 LuCI 接口统计溢出显示；当前 LuCI 已使用新版 JavaScript 状态页面和可处理大计数器的格式化逻辑，旧版负数显示问题已被后续实现覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf