coolsnowwolf/lede #8243: 最近编译固件里的DDNS工作不正常

wangshengxin1998, 2021-11-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：cloudflare.com-v4 DDNS脚本不能正常更新根域名的记录,子域名可以正常更新

### (2) 路由器型号和固件版本 (无关)

A：Nanopi R4S

### (3) 详细日志

```
 180110       : ************ ************** ************** **************
 180110  note : PID '13941' started at 2021-11-12 18:01:10
 180110       : ddns version  : 2.7.8-3
 180110       : uci configuration:
ddns.cloudflare1.cacert='IGNORE'
ddns.cloudflare1.check_interval='5'
ddns.cloudflare1.domain='@domain.xyz'
ddns.cloudflare1.enabled='1'
ddns.cloudflare1.force_interval='96'
ddns.cloudflare1.interface='wan'
ddns.cloudflare1.ip_interface='pppoe-wan'
ddns.cloudflare1.ip_source='interface'
ddns.cloudflare1.lookup_host='domain.xyz'
ddns.cloudflare1.password='***PW***'
ddns.cloudflare1.retry_interval='30'
ddns.cloudflare1.service_name='cloudflare.com-v4'
ddns.cloudflare1.use_https='1'
ddns.cloudflare1.use_syslog='3'
ddns.cloudflare1.username='email@addr.tld'
ddns.cloudflare1=service
 180112       : verbose mode  : 0 - run normal, NO console output
 180112       : check interval: 300 seconds
 180112       : force interval: 345600 seconds
 180113       : retry interval: 30 seconds
 180113       : retry counter : 0 times
 180113       : No old process
 180113       : last update: never
 180113       : Detect registered/public IP
 180113       : #> /usr/bin/nslookup domain.xyz  >/var/run/ddns/cloudflare1.dat 2>/var/run/ddns/cloudflare1.err
 180114       : Registered IP '3.4.5.6' detected
 180114  info : Starting main loop at 2021-11-12 18:01:14
 180114       : Detect local IP on 'interface'
 180114       : #> ip -o addr show dev pppoe-wan scope global >/var/run/ddns/cloudflare1.dat 2>/var/run/ddns/cloudflare1.err
 180114       : Local IP '1.2.3.4' detected on interface 'pppoe-wan'
 180115       : Update needed - L: '1.2.3.4' <> R: '3.4.5.6'
 180115       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 180115       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflare1.dat --stderr /var/run/ddns/cloudflare1.err --insecure --noproxy '*' --header 'X-Auth-Email: email@addr.tld'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=domain.xyz'
 180118       : #> /usr/bin/curl -RsS -o /var/run/ddns/cloudflare1.dat --stderr /var/run/ddns/cloudflare1.err --insecure --noproxy '*' --header 'X-Auth-Email: email@addr.tld'  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones/ad39776480adfaffc6fc5acc1790f54e/dns_records?name=@.domain.xyz&type=A'
 180122  WARN : Could not detect 'record id' for host.domain.tld: '@.domain.xyz'
 180122 ERROR : No update send to DDNS Provider
 180122       : Waiting 300 seconds (Check Interval)
```

wangshengxin1998, 2021-11-12:
我暂时从较旧的固件里提取相应的脚本覆盖到相应的位置来暂时解决这个问题

xeyesen, 2021-11-18:
我的也是

wangshengxin1998, 2021-11-18:
> 我的也是

需要单独提供旧版的脚本吗?

suizhenmin, 2021-11-19:
原来旧版是可以的啊，我以为是不支持。

jerryzhujian9, 2021-11-25:
> 反馈bug/问题模板，提建议请删除
> ## 1.关于你要提交的问题
>
> Q：是否搜索了issue (使用 "x" 选择)
>
>     * [x]  没有类似的issue
>
>
> ## 2. 详细叙述
> ### (1) 具体问题
>
> A：cloudflare.com-v4 DDNS脚本不能正常更新根域名的记录,子域名可以正常更新
> ### (2) 路由器型号和固件版本 (无关)
>
> A：Nanopi R4S [...]

我装了ddns-scripts-cloudflare,但是在luci的ddns界面中显示不出来,你是咋弄来的? @wangshengxin1998

closed 2021-12-05 by wangshengxin1998

tiantaichenhui, 2022-07-08:
> 暂时从较旧的收费里，我会相应地覆盖到相应的位置来解决这个问题

遇到同样的问题了 请问脚本在什么路径？我也准备覆盖下试试

Kissycat, 2023-01-29:
2年前制造的bug，在两年前[bug产生](https://github.com/coolsnowwolf/packages/commit/d51b7bd999860530d23103564fc314158915396b)的第二天就被人发现了，结果两年后的今天还没修复，受害用户+1🥲