coolsnowwolf/lede #8247: 大佬能否在ARM机器上编译本源码

sw586, 2021-11-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
在执行sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync
时报错：

```
root@instance-20211112-1458:~# sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync
Reading package lists... Done
Building dependency tree... 50%
Building dependency tree
Reading state information... Done
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'git' instead of 'git-core'
Note, selecting 'upx-ucl' instead of 'upx'
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package lib32gcc1
E: Unable to locate package libc6-dev-i386
E: Package 'gcc-multilib' has no installation candidate
E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'
```

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：我知道报错是构架不对，想问下大佬是否有ARM上编译的方案

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

LPDDR6-10000MHz, 2021-11-13:
[image]

sw586, 2021-11-13:
这么大的大佬亲和力这么好阿  至少来两个字嘛

minlang112, 2021-11-14:
不要浪费时间

sw586, 2021-11-14:
> 不要浪费时间

好的 大佬

liwenjie119, 2021-11-15:
https://www.right.com.cn/forum/thread-4057579-1-1.html
参考这个，之前我用手机编译，最近没有编译了，不知道有没有新的问题出现

WYC-2020, 2021-11-17:
手机编译 哈哈，想象力挺丰富

closed 2021-11-30 by AmadeusGhost