coolsnowwolf/lede #8252: 32位编译，netdata无法工作，显示错误

dfbucy090, 2021-11-14:
昨天看到toolchain升级到toolchain/musl: update to 1.2.2，用32位又编译了一遍，结果bash可以用，但是插件netdata无法工作，显示错误。具体是

192.168.1.2:19999上的嵌入式页面显示
ERROR 405: Cannot download charts index from server: http://192.168.1.2:19999/api/v1/charts

toolchain升级前编译，bash不能用。后来有大佬帮助给了一组代码，Lede目录先前执行代码然后再编译，bash可用，但是和这次升级结果一样，netdata插件不能用。如果先编译再执行代码然后再编译一遍，就比较完美，netdata插件和bash都可以用。

本人小白，希望得到大佬帮助

dfbucy090, 2021-11-24:
今天5.4.161版本又编译了一遍，netdata插件还是弹出错误显示
192.168.1.2：19999上的嵌入式页面显示
Cannot load required JS library:http://192.168.1.2:19999/lib/
perfect-scrollbar-0.6.15.min.js

WYC-2020, 2021-11-25:
32位一堆异常基本都不用了，
[image]

WYC-2020, 2021-11-25:
这个musl升级到1.2.2后那个time_t的定义已经变了，有的库在格式化的时候会出问题，我不知道这些异常是不是就是由于这个原因导致的，后面慢慢看吧

WYC-2020, 2021-11-26:
看了一波这些错误都是netdata的，感觉是netdata的插件和musl1.2.2匹配不好

WYC-2020, 2021-11-30:
已修复，证实是我猜测的问题，netdata对时间格式化错误，改下就可以了

dfbucy090, 2021-12-01:
> 已修复，证实是我猜测的问题，netdata对时间格式化错误，改下就可以了
我刚才又编译了一次，还是不行，点击netdata时，显示白板，中间提示192.168.1.2拒绝我们的连接请求。编译时是还要执行或更改什么吗？

WYC-2020, 2021-12-02:
@dfbucy090 我没提交pr，自己进我空间package去看，我改动的包很多所以没有提pr，自己找自己需要按我哪个修改即可

dfbucy090, 2021-12-02:
> @dfbucy090 我没提交pr，自己进我空间package去看，我改动的包很多所以没有提pr，自己找自己需要按我哪个修改即可

您好，看到您的有关netdata的0001-fix-netdata-for-musl-1.2.x.patch有更改，可是不懂怎么操作，我只会按lede教程编译，如果有更改，具体怎么弄就不会了，太小白了。不过还是谢谢您的关注。

WYC-2020, 2021-12-02:
@dfbucy090 我晕不会？真小白，把netdata全部复制过来替换相同就行，或者直接用我那个package，还有就是等netdata官方更新，我已经给他们说了，可能下个版本就会修复，这个是一个补丁

dfbucy090, 2021-12-03:
> @dfbucy090 我晕不会？真小白，把netdata全部复制过来替换相同就行，或者直接用我那个package，还有就是等netdata官方更新，我已经给他们说了，可能下个版本就会修复，这个是一个补丁

不好意思，真没学过语言，只是看视频照葫芦画瓢，我理解是，编译时是在执行git clone https://github.com/coolsnowwolf/lede之前或者之后，然后，复制您的文件夹过来覆盖lede的，具体怎么操作就不会了。然后再按教程步骤进行下去。

realconnor, 2021-12-15:
似乎是最近更新到netdata 1.31，出现了奇怪的问题，如图
[image]
会闪现1秒正常的仪表界面，然后就变成空白，以前的版本没有出现过这个问题，求修复。

WYC-2020, 2021-12-15:
@realconnor 没用的他们还没修复，我提交的补丁他们都没更新，我自己已经修复了

dfbucy090, 2021-12-15:
> 似乎是最近更新到netdata 1.31，出现了奇怪的问题，如图 [image] 会闪现1秒正常的仪表界面，然后就变成空白，以前的版本没有出现过这个问题，求修复。

我也发现了这个问题，还是不会正确编译，您自己修复了，能具体告诉我怎么操作吗？好比怎么把您的修复文件复制过来，然后正确编译，具体到执行指令。

realconnor, 2021-12-16:
没修复，我平时也很少看仪表盘，偶然发现这了问题。平时都是云编译的，应该是补丁和netdata新版本不兼容。发自我的华为手机-------- 原始邮件 --------发件人： dfbucy090 ***@***.***>日期： 2021年12月16日周四 07:22收件人： coolsnowwolf/lede ***@***.***>抄送： realconnor ***@***.***>, Mention ***@***.***>主    题： Re: [coolsnowwolf/lede] 32位编译，netdata无法工作，显示错误 (Issue #8252)

似乎是最近更新到netdata 1.31，出现了奇怪的问题，如图  会闪现1秒正常的仪表界面，然后就变成空白，以前的版本没有出现过这个问题，求修复。

我也发现了这个问题，还是不会正确编译，您自己修复了，能具体告诉我怎么操作吗？好比怎么把您的修复文件复制过来，然后正确编译，具体到执行指令。

WYC-2020, 2021-12-16:
刚看了 确实升级到1.31后会出现这个空白问题，这个应该不属于time_t造成的了，应该是那里有坑，不升级用上一个版本，如果是musl1.2.x话，加入我哪个补丁应该就可以用了

WYC-2020, 2021-12-16:
看了追新是不好的，不要追新了哈哈

WYC-2020, 2021-12-16:
这应该和1.31改了web布局的关系，我看了netdata-1.31.0/web/gui下面已经大变样了，再old文件夹下面才是老的了

WYC-2020, 2021-12-16:
@dfbucy090  @realconnor  我发现我清除浏览器缓存就可以了，你们试下

realconnor, 2021-12-16:
没用，这种操作当然都试过了，才来提的问题…

WYC-2020, 2021-12-16:
[image]
看这个错误应该是js脚本那里出粗了，我试过还原到1.30.1一样的

realconnor, 2021-12-16:
加个速，你试试最新的 [v1.32.1](https://github.com/netdata/netdata/releases/tag/1.32.1) 看看？不知道这破玩意升级它干什么，官方的 [软件源](https://github.com/openwrt/packages/blob/master/admin/netdata/Makefile) 还是v1.30.1，前段时间莫名默认加了个docker也是，简直是偷袭😥 各种代理/防火墙转发都出问题。我怀疑有些“开发者”发布的commits甚至都没自己测试过…

其实最省心的办法还是在官方稳定源码上打补丁，也方便给上游发pr。现在这种内核和某些软件追新，但是Web仍然是18.06的模式 固然有兼容的考虑，但是今后问题只会越来越多。

WYC-2020, 2021-12-16:
我怀疑是更新了什么导致的，这个我修复过一次的，最近我跟到l大更新了库，你刚说有问题我编译了一下还真是这样，不知道是哪个导致的，哎，我试过1.32.1一样的，不知道什么鬼不修了，难得搞，反正我也不用，大不了不编译这玩意

WYC-2020, 2021-12-16:
不单32位这次连64位我这边测试也是一样

WYC-2020, 2021-12-17:
简单搞了一下，我这边测试没问题
[image]
https://github.com/WYC-2020/packages/commit/ffab2428fc468d493d6fce762aa9d95122f7d8e4 @realconnor

realconnor, 2021-12-17:
> 简单搞了一下，我这边测试没问题
> [image]
> https://github.com/WYC-2020/packages/commit/ffab2428fc468d493d6fce762aa9d95122f7d8e4 @realconnor 

你应该发pr给上游…
我要这玩意也没用，直接删掉了😂

dfbucy090, 2021-12-17:
> 简单搞了一下，我这边测试没问题 [image] [WYC-2020/packages@ffab242](https://github.com/WYC-2020/packages/commit/ffab2428fc468d493d6fce762aa9d95122f7d8e4) @realconnor

看着很好，可惜我看不懂，也不会弄，你应该提交给舵主，让他们改正，我们只需按原来操作就能编译成功，这样才叫功德圆满。谢谢您！

dfbucy090, 2021-12-23:
> 简单搞了一下，我这边测试没问题 [image] [WYC-2020/packages@ffab242](https://github.com/WYC-2020/packages/commit/ffab2428fc468d493d6fce762aa9d95122f7d8e4) @realconnor

谢谢您，用您的修改的解决了。

dfbucy090, 2021-12-28:
> 简单搞了一下，我这边测试没问题 [image] [WYC-2020/packages@ffab242](https://github.com/WYC-2020/packages/commit/ffab2428fc468d493d6fce762aa9d95122f7d8e4) @realconnor

昨天看到您提交了一个新的netdata帖子，怎么又收回了？怎么知道什么时候不用您的补丁，可以直接编译了呢？

WYC-2020, 2021-12-28:
@dfbucy090 我记得我没提交任何netdata的帖子呢，那是别人提的吧，补丁是更加netdata的版本来如果下一个版本以及修复就可以去掉补丁

dfbucy090, 2021-12-28:
> @dfbucy090 我记得我没提交任何netdata的帖子呢，那是别人提的吧，补丁是更加netdata的版本来如果下一个版本以及修复就可以去掉补丁

昨天看到有个Issues是您发的，说是有几个人提出netdata问题，您说您已经解决了，还给出了链接，然后说你们大家谁愿意提交一下就可以了。今天看，这个Issues就没有了。我还是不知道啥时候撤销补丁，不懂哪个版本已经更新或者修复。

WYC-2020, 2021-12-28:
@dfbucy090 不是我提的那个，是别人提的，我给了解决方案，他就关了

dfbucy090, 2021-12-28:
已经解决

coolsnowwolf, 2026-09-06:
该报告针对 2021 年 32 位旧构建环境，只有 HTTP 405 现象，没有可复现的 netdata 配置、版本和日志。当前 netdata/toolchain 已多次更新，无法将旧环境现象归因到当前仓库缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf