coolsnowwolf/lede #8257: 这个报错好几天了，有没有大佬知道啥原因

MXJNZ6, 2021-11-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：pppd和pptpd的报错是一个么，2、3天了都没看人提，昨天见一个pull提ppp的但是他说他第一次提，水平也不行，又撤回了。

### (2) 路由器型号和固件版本

A：armv8-a53

### (3) 详细日志

A：
pptpd-logwtmp.c: In function 'ip_up':
pptpd-logwtmp.c:47:5: warning: implicit declaration of function 'notice'; did you mean 'nice'? [-Wimplicit-function-declaration]
     notice("pptpd-logwtmp.so ip-up %s %s %s", ifname, user,
     ^~~~~~
     nice
pptpd-logwtmp.c:49:3: warning: implicit declaration of function 'logwtmp'; did you mean 'updwtmp'? [-Wimplicit-function-declaration]
   logwtmp(ifname, user, pptpd_original_ip);
   ^~~~~~~
   updwtmp
pptpd-logwtmp.c: In function 'plugin_init':
pptpd-logwtmp.c:61:3: warning: implicit declaration of function 'add_options' [-Wimplicit-function-declaration]
   add_options(options);
   ^~~~~~~~~~~
pptpd-logwtmp.c:62:3: warning: implicit declaration of function 'add_notifier' [-Wimplicit-function-declaration]
   add_notifier(&ip_up_notifier, ip_up, NULL);
   ^~~~~~~~~~~~
上边的那样的报错很长
make[6]: *** [pptpd-logwtmp.so] Error 1
make[5]: *** [all-local] Error 2
make[4]: *** [all] Error 2
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pptpd-1.4.0/.built] Error 2
make[2]: *** [package/feeds/packages/pptpd/compile] Error 1
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2
Makefile:19: recipe for target 'pptpd-logwtmp.so' failed
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pptpd-1.4.0/plugins'
Makefile:840: recipe for target 'all-local' failed
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pptpd-1.4.0'
Makefile:277: recipe for target 'all' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pptpd-1.4.0'
Makefile:69: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pptpd-1.4.0/.built' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/pptpd'
time: package/feeds/packages/pptpd/compile#0.17#0.05#0.21
    ERROR: package/feeds/packages/pptpd failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/pptpd/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed

souwei168, 2021-11-15:
同样问题
我还以为插件问题

jingleijack, 2021-11-15:
pptp插件删除可以解决

wj45853007, 2021-11-15:
同样问题 带PPTP 过不了，看会不会解决...

coolsnowwolf, 2021-11-16:
fixed

closed 2021-11-16 by coolsnowwolf