coolsnowwolf/lede #8258: 这个用GITHUB 自动编译，之前没有问题，最近几天都报错无法编译！

souwei168, 2021-11-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
这个用GITHUB 自动编译，之前没有问题，最近几天都报错无法编译！
### (1) 具体问题
这个用GITHUB 自动编译，之前没有问题，最近几天都报错无法编译！
A：

### (2) 路由器型号和固件版本
X86
A：

### (3) 详细日志
install -m0755 /workdir/openwrt/dl/qbittorrent-nox /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static/usr/bin
find /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="true" STRIP_KMOD="/workdir/openwrt/scripts/strip-kmod.sh" PATCHELF="/workdir/openwrt/staging_dir/host/bin/patchelf" /workdir/openwrt/scripts/rstrip.sh /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static
rstrip.sh: /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static/usr/bin/qbittorrent-nox: executable
patchelf: cannot find section '.dynamic'. The input file is most likely statically linked
(cd /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/base
/workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static /workdir/openwrt/bin/packages/x86_64/base
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/qBittorrent-static-4.3.9_v1.2.14/ipkg-x86_64/qBittorrent-static into /workdir/openwrt/bin/packages/x86_64/base/qBittorrent-static_4.3.9_v1.2.14-1_x86_64.ipk
echo "qBittorrent-static" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/qBittorrent-static.default.install
make[3]: Leaving directory '/workdir/openwrt/package/lean/qBittorrent-static'
time: package/lean/qBittorrent-static/compile#2.62#0.34#2.98
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

A：

jingleijack, 2021-11-15:
我也是遇到了。删除插件PPTP解决了。但是昨天有出现编译失败

jingleijack, 2021-11-15:
你的需要删除qBittorrent插件即可成功

souwei168, 2021-11-15:
我看报错也是QB这个插件的问题，但是我想要保留这个插件，有什么办法没有
，比如我再拉一个第三方的QB源，放在LEAN的上级目录，不知可行吗？

tianiue, 2021-11-16:
没有编译QB，也会报错

souwei168, 2021-11-16:
确实，总之怎么搞，都无法自动编译了，大佬，不知道改动了哪里？

souwei168, 2021-11-16:
(cd /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-poweroffdevice/ipkg-all/luci-app-poweroffdevice/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/base
/workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-poweroffdevice/ipkg-all/luci-app-poweroffdevice /workdir/openwrt/bin/packages/x86_64/base
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/nps-0.26.10/.pkgdir/npc/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.npc_installed
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/luci-app-poweroffdevice/ipkg-all/luci-app-poweroffdevice into /workdir/openwrt/bin/packages/x86_64/base/luci-app-poweroffdevice_1-3 _all.ipk
bash: line 0: [: /workdir/openwrt/bin/packages/x86_64/base/luci-app-poweroffdevice_1-3: binary operator expected
make[3]: *** [/workdir/openwrt/feeds/luci/luci.mk:255: /workdir/openwrt/bin/packages/x86_64/base/luci-app-poweroffdevice_1-3] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/luci-app-poweroffdevice'
time: package/lean/luci-app-poweroffdevice/compile#0.24#0.08#0.37
    ERROR: package/lean/luci-app-poweroffdevice failed to build.
make[2]: *** [package/Makefile:116: package/lean/luci-app-poweroffdevice/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
echo "npc" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/nps.default.install
make[3]: Leaving directory '/workdir/openwrt/package/lean/nps'
time: package/lean/nps/compile#0.28#0.08#0.39
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

closed 2021-11-16 by coolsnowwolf