coolsnowwolf/lede #8260: x86固件 sftp翻车

xiangfeidexiaohuo, 2021-11-15:
Mon Nov 15 15:49:32 2021 authpriv.info dropbear[30553]: Child connection from 192.168.2.11:49814
Mon Nov 15 15:49:35 2021 authpriv.notice dropbear[30553]: Password auth succeeded for 'root' from 192.168.2.11:49814
Mon Nov 15 15:49:35 2021 authpriv.info dropbear[30561]: Child connection from 192.168.2.11:49820
Mon Nov 15 15:49:35 2021 authpriv.notice dropbear[30561]: Password auth succeeded for 'root' from 192.168.2.11:49820
Mon Nov 15 15:49:36 2021 kern.info kernel: [ 6012.431797] sftp-server[30573]: segfault at 53ed8070 ip 000055ae96d0a711 sp 00007ffe375ee9f8 error 6 in sftp-server[55ae96d0a000+11000]
Mon Nov 15 15:49:36 2021 kern.info kernel: [ 6012.442995] Code: c4 88 04 00 00 5b 5d 41 5c 41 5d 41 5e 41 5f c3 8b 15 73 a9 01 00 48 63 c7 89 3d 6a a9 01 00 48 6b c0 38 48 03 05 57 ab 01 00 <c7> 00 00 00 00 00 89 50 30 c3 31 c0 85 ff 78 1d 3b 3d 39 ab 01 00
Mon Nov 15 15:49:37 2021 authpriv.info dropbear[30561]: Exit (root) from <192.168.2.11:49820>: Exited normally
Mon Nov 15 15:49:37 2021 authpriv.info dropbear[30611]: Child connection from 192.168.2.11:49821
Mon Nov 15 15:54:37 2021 authpriv.info dropbear[30611]: Exit before auth from <192.168.2.11:49821>: Timeout before auth

能登录ssh，但是访问不了sftp，昨天上午还没事，今天更新，重新编译出固件，就访问不了

xiangfeidexiaohuo, 2021-11-15:
5.10.79内核问题，降级到 5.10.78就可以了

xiangfeidexiaohuo, 2021-11-15:
5.10分支的  降回78的时候 刚重启 没事，过一段时间，还是依旧，很奇怪

Beginner-Go, 2021-11-15:
最近没动sftp 不应该源码问题

xiangfeidexiaohuo, 2021-11-15:
> 最近没动sftp 不应该源码问题

我也搞不懂了  昨天上午十点左右pull源码 编译的 没问题，后面再编译的 就一直有问题，不管是5.4 还是 5.10内核，我也很纳闷

xiangfeidexiaohuo, 2021-11-15:
找到原因，docker-compose 我手动升级到 v2.1.1版本了，退回到 v2.1.0 或者 1.29.2都成

WYC-2020, 2021-11-16:
我是直接全部失败，不管ssh还是sftp，
但是如果你调试模式运行就能登录，但是只能登录一个，并且你断开的时候，ssh会自动退出，佛学不懂哈哈

WYC-2020, 2021-11-16:
@xiangfeidexiaohuo  
https://stackfault.net/t/topic/46 这里有一个人也遇到了 和我这个症状简直一模一样，应该解决那个31信号应该就能解决了

xiangfeidexiaohuo, 2021-11-16:
> @xiangfeidexiaohuo https://stackfault.net/t/topic/46 这里有一个人也遇到了 和我这个症状简直一模一样，应该解决那个31信号应该就能解决了

我是刚重启后 那一会儿 可登录sftp 然后就不行，ssh倒是没问题

WYC-2020, 2021-11-16:
那可能是你那里冲突了吧，我这边和你症状不同 @xiangfeidexiaohuo

WYC-2020, 2021-11-16:
@xiangfeidexiaohuo  刚我试了下 可以，我原来一直想通过openssh 组合openssh-sftp-server 来搭建sftp。我刚试了下直接dropbear+openssh-sftp-server 搭建没问题的，openssh那个31信号那个确实是一个问题，我已经提了一个issue了。sftp我这边按我这个组合没问题，不知道你是怎么组合的

xiangfeidexiaohuo, 2021-11-16:
> dropbear

删除openssh-sftp-server可以这样连接
[image]

但是不能单独sftp访问
[image]

WYC-2020, 2021-11-16:
我的都可以 只要那样组合

WYC-2020, 2021-11-16:
[image]
我这边ftp sftp 已经finalshell都可以

xiangfeidexiaohuo, 2021-11-16:
> 我的都可以 只要那样组合

你有哪些sftp相关的插件 在固件里

WYC-2020, 2021-11-16:
[image] 
就这个，我自己修改了vsftp增加了开启开关，就openssh-sftp-server这个

xiangfeidexiaohuo, 2021-11-16:
我要安装openssh-sftp-server | 8.8p1-3 就能用，问题是 这个 -3版本的源码在哪呢

WYC-2020, 2021-11-16:
-3版本？我用的都是默认lede的好像是
[image]
如果-3你的可以用你升级一下不就行了，修改makefile,这个源码makefile里面不是有地址PKG_SOURCE_URL:=https://ftp.openbsd.org/pub/OpenBSD/OpenSSH/portable/ \
                https://ftp.spline.de/pub/OpenBSD/OpenSSH/portable/ 自己去找有没-3吧 我是用默认的就可以的

WYC-2020, 2021-11-16:
@xiangfeidexiaohuo 要的-3 我怀疑是不是这里https://github.com/openssh/openssh-portable 的v_8_3,自己去看下吧

xiangfeidexiaohuo, 2021-11-16:
> -3版本？我用的都是默认lede的好像是 [image] 如果-3你的可以用你升级一下不就行了，修改makefile,这个源码makefile里面不是有地址PKG_SOURCE_URL:=https://ftp.openbsd.org/pub/OpenBSD/OpenSSH/portable/ https://ftp.spline.de/pub/OpenBSD/OpenSSH/portable/ 自己去找有没-3吧 我是用默认的就可以的

这是确实没，但是 在线仓库里 确实有个-3 
[image]

WYC-2020, 2021-11-16:
那不清楚了 你这个仓库是lede官方提供的仓库，还是某人自己编译的，如果是某人自己编译的可能是别人升级过，然后编译出来的，我的那个版本是了的官方提供源码所带的版本 @xiangfeidexiaohuo

xiangfeidexiaohuo, 2021-11-16:
> 那不清楚了 你这个仓库是lede官方提供的仓库，还是某人自己编译的，如果是某人自己编译的可能是别人升级过，然后编译出来的，我的那个版本是了的官方提供源码所带的版本 @xiangfeidexiaohuo

这个仓库就是 Lean
自带的插件仓库

WYC-2020, 2021-11-16:
那就不清楚了，从开放的源码看openssh版本还没到3

greennyreborn, 2021-11-17:
[#](https://github.com/coolsnowwolf/packages/issues/286)
我最近编译的版本，openssh sshd启动不了，包segfault，和你们说的问题是不是有关联

WYC-2020, 2021-11-17:
我昨天也遇到，但是全新编译后没这个问题 @greennyreborn

breakings, 2021-11-17:
这个问题出现很久了，目前GCC11编译出来，openssh组件都无法使用，官方源的就完全正常。

WYC-2020, 2021-11-17:
不会啊，我就是最新的gcc编译的能用啊

xiangfeidexiaohuo, 2021-11-17:
> 这个问题出现很久了，目前GCC11编译出来，openssh组件都无法使用，官方源的就完全正常。

sftp 这个 我目前 不默认编译的 ，要用就在线装，在线没问题。至于纯ssh，没啥问题。

breakings, 2021-11-18:
> > 这个问题出现很久了，目前GCC11编译出来，openssh组件都无法使用，官方源的就完全正常。
> 
> sftp 这个 我目前 不默认编译的 ，要用就在线装，在线没问题。至于纯ssh，没啥问题。

我这边还是不行，无法通过openssh连接ssh和sftp，client_loop: send disconnect: Broken pipe

WYC-2020, 2021-11-18:
我都ok

Neverends4, 2021-11-19:
某些ssh client对op的dropbear支持不好，如果不装openssh-sftp-server就没法用sftp连接，如果你只是sftp没法连接，ssh可以
就改用scp连接，界面和效果与sftp类似
支持scp连接的win客户端有winscp和filezilla

greennyreborn, 2021-11-19:
> 我都ok

我全新编译完还是有问题，sshd 可以启动了，但是客户端连不上，会直接断开。
请问你编译有选了什么特别的选项么

WYC-2020, 2021-11-19:
@greennyreborn  那个我不是发了issue 然后我找到解决办法，不知道你这个是不是和我一样的症状，我加入这个选项解决--with-sandbox=rlimit

greennyreborn, 2021-11-19:
> @greennyreborn 那个我不是发了issue 然后我找到解决办法，不知道你这个是不是和我一样的症状，我加入这个选项解决--with-sandbox=rlimit

哪个issue，能麻烦给个链接么
还有这个 ```--with-sandbox=rlimit```是直接加到openssh的makefile里的么

WYC-2020, 2021-11-19:
https://github.com/coolsnowwolf/lede/issues/8262#issue-1054383232

WYC-2020, 2021-11-19:
我这边改了直接编译出来的包就开启 dro和openssh 两个同时开启哈哈，ftp和sftp 也是一样

breakings, 2021-11-21:
make dirclean 大法好 重新编译就都能用了

warry256, 2021-11-27:
我编译的R2S固件（无处密码），ttyd终端和tftp可以登陆，加上管理员密码后都不能用了

coolsnowwolf, 2026-09-06:
当前 packages 的 openssh 已更新到 10.0p2，sftp-server 使用独立的新版 OpenSSH 实现；报告中的 2021 年 sftp-server 崩溃无法在现行版本复现，旧版本问题已被更新覆盖，关闭为 completed。

closed 2026-09-06 by coolsnowwolf