coolsnowwolf/lede #8261: pptpd failed to build.

smallprogram, 2021-11-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：pptpd failed to build.

### (2) 路由器型号和固件版本

A：X86
kernel: 5.4.158 

### (3) 详细日志
```
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:670:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifvjcomp __P((int, int, int, int));
                ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:672:12: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifup __P((int));  /* Configure i/f up for one protocol */
            ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:673:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifnpmode __P((int u, int proto, enum NPmode mode));
                ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:675:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifdown __P((int)); /* Configure i/f down for one protocol */
              ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:676:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifaddr __P((int, u_int32_t, u_int32_t, u_int32_t));
              ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:678:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  cifaddr __P((int, u_int32_t, u_int32_t));
              ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:689:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifdefaultroute __P((int, u_int32_t, u_int32_t, bool replace_default_rt));
                      ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:691:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  cifdefaultroute __P((int, u_int32_t, u_int32_t));
                      ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:699:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  sifproxyarp __P((int, u_int32_t));
                  ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:701:18: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  cifproxyarp __P((int, u_int32_t));
                  ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:703:19: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 u_int32_t GetMask __P((u_int32_t)); /* Get appropriate netmask for address */
                   ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:704:11: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  lock __P((char *)); /* Create lock file for device */
           ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:705:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  relock __P((int));  /* Rewrite lock file with new pid */
             ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:706:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 void unlock __P((void)); /* Delete previously-created lock file */
             ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:707:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 void logwtmp __P((const char *, const char *, const char *));
              ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:709:20: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  get_host_seed __P((void)); /* Get host-dependent random number seed */
                    ^~~
/workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/usr/include/pppd/pppd.h:710:20: error: expected '=', ',', ';', 'asm' or '__attribute__' before '__P'
 int  have_route_to __P((u_int32_t)); /* Check if route to addr exists */
[... 114 lines cut ...]
   updwtmp
pptpd-logwtmp.c: In function 'plugin_init':
pptpd-logwtmp.c:61:3: warning: implicit declaration of function 'add_options' [-Wimplicit-function-declaration]
   add_options(options);
   ^~~~~~~~~~~
pptpd-logwtmp.c:62:3: warning: implicit declaration of function 'add_notifier' [-Wimplicit-function-declaration]
   add_notifier(&ip_up_notifier, ip_up, NULL);
   ^~~~~~~~~~~~
make[6]: *** [Makefile:19: pptpd-logwtmp.so] Error 1
make[5]: *** [Makefile:840: all-local] Error 2
make[4]: *** [Makefile:277: all] Error 2
make[3]: *** [Makefile:72: /workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/pptpd-1.4.0/.built] Error 2
make[2]: *** [package/Makefile:116: package/feeds/packages/pptpd/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/pptpd-1.4.0/plugins'
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/pptpd-1.4.0'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a72_musl/pptpd-1.4.0'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/pptpd'
time: package/feeds/packages/pptpd/compile#0.22#0.10#0.31
    ERROR: package/feeds/packages/pptpd failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
Error: Process completed with exit code 2.
```

abrahkadzabbrah, 2021-11-15:
many also experience it.. temporarily remove packages related to `ppp` so that compilation runs smoothly, while waiting for someone to fix it

hzjnet, 2021-11-15:
应该是内核升级后对应补丁未提供吧，出来好几天了！

LPDDR6-10000MHz, 2021-11-16:
https://github.com/coolsnowwolf/lede/pull/8263

coolsnowwolf, 2021-11-16:
fixed

closed 2021-11-16 by coolsnowwolf