coolsnowwolf/lede #8262: openssh 因“client_loop: send disconnect: Broken pipe”而失败，但在调试模式下工作

WYC-2020, 2021-11-16:
我正在尝试在我的 openwrt 路由器上运行 openssh。为了测试它，我通过/usr/sbin/sshd -D启动 openssh。系统日志显示：
Tue Nov 16 10:59:41 2021 auth.info sshd[11926]: Accepted password for root from 192.168.110.1 port 1858 ssh2
Tue Nov 16 10:59:41 2021 auth.crit sshd[11926]: fatal: privsep_preauth: preauth child terminated by signal 31

现在我尝试调试问题，但是当我通过/usr/sbin/sshd -D -d在调试模式下运行 openssh 时，连接正常工作并且没有遇到错误！

可能是什么问题以及如何解决它？

WYC-2020, 2021-11-16:
搞定，增加一个选项--with-sandbox=rlimit 就可以了，这个bug官方又说，说是修复了，但是不知咋的在op上面会在此出现，加入这个选项就可以避免这个情况，关闭了

closed 2021-11-16 by WYC-2020

v03413, 2022-01-08:
openssh 在openwrt 官方仓库上面编译不会出现此问题，至于lede为啥会出现 原因不明。  
---  

楼上老哥已经给出了具体方法，我来说下具体操作步骤：
修改文件：`编译目录/package/feeds/packages/openssh/Makefile`，搜索关键词：`CONFIGURE_ARGS`，然后按照相同格式追加一个选项`--with-sandbox=rlimit`，然后重新编译，会发现已经正常啦，截图如下：

[image]

BecodReyes, 2022-06-18:
补充一下，[2a88d03](https://github.com/coolsnowwolf/packages/commit/2a88d0391623742de82cdebd1d9c586abec8e391)的修复修复了个寂寞，漏了反斜杠

补上会出现`rlimit sandbox requires select to work with rlimit`

需要回退commit，目前测试到可用的是[def4f1a](https://github.com/coolsnowwolf/packages/commit/def4f1ab3cfb342d60807447739b2475c3db666c)

dramenk, 2022-06-23:
交叉编译的时候 select_works_with_rlimit 的检查项被默认置为 no 了，比较奇怪这个选项是怎么加上去不报错的

CHxCOOH, 2022-09-10:
补充：我在新版固件里又遇到了同样问题。解决方案：安装和openssh版本匹配的openssl：`opkg install libopenssl-conf libopenssl1.1 openssl-util`。注意：软件源中的openssl版本可能高于openssh版本，请自行解决（手动升级openssh（P.s.比这新的正式版还没发）或者手动降级openssl）。经过测试，openssh版本8.9/9.0p1-1可以对应openssl1.1.1q-1，而更高版本的openssl1.1.1q-6是不可用的。

具体做法：先用sftp/web文件管理上传一份1.1.1q-1的ipk，可以在[清华镜像站](http://mirrors.tuna.tsinghua.edu.cn/openwrt/releases/22.03.0/packages/aarch64_generic/base/)下载，需要上述三个包，存下来传到路由器上。然后`rm /var/opkg-lists/*`删除版本信息，否则包管理器会自动下载更新的版本。然后卸载高版本：
```sh
opkg --force-depends remove libopenssl1.1
opkg --force-depends remove libopenssl-conf
opkg --force-depends remove openssl-util
```
（如果熟练的话也可以尝试直接覆盖安装，使用--force-downgrade）
再`opkg install`之前上传好的三个包就行了。