coolsnowwolf/lede #8270: MF269 web升级保留配置升级后，无线启动不了

lgh200, 2021-11-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

[A：](url)
[image]
[image]

### (2) 路由器型号和固件版本

A：MF269

### (3) 详细日志

A：系统日志
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Wed Nov 17 08:21:02 2021 kern.notice kernel: [    0.000000] Linux version 5.10.79 (lgh@DESKTOP-1GU35TK) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r3606-f5c0cee30) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP Tue Nov 16 12:33:55 2021
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Machine model: ZTE MF269
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Zone ranges:
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   DMA32    empty
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   Normal   empty
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Early memory node ranges
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a5fffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e7fffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004e800000-0x00000000504fffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000050500000-0x00000000506fffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000050700000-0x000000005fffffff]
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 131072
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000]   DMA zone: 2048 pages used for memmap
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000]   DMA zone: 131072 pages, LIFO batch:31
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] percpu: Embedded 20 pages/cpu s41112 r8192 d32616 u81920
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s41112 r8192 d32616 u81920 alloc=20*4096
Wed Nov 17 08:21:02 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Wed Nov 17 08:21:02 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] Memory: 416336K/524288K available (7230K kernel code, 838K rwdata, 1868K rodata, 1280K init, 300K bss, 107952K reserved, 0K cma-reserved)
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Wed Nov 17 08:21:02 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x350/0x570 with crng_init=0
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000005] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Wed Nov 17 08:21:02 2021 kern.info kernel: [    0.000156] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=76800)
[... 734 lines cut ...]
[  139.505507] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[  139.507112] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[  139.512600] kmodloader: done loading kernel modules from /etc/modules.d/*
[  139.512638] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0x292 soc_id 0xffffffff
[  139.524857] ath11k c000000.wifi: fw_version 0x250684a5 fw_build_timestamp 2021-07-13 10:57 fw_build_id QC_IMAGE_VERSION_STRING=WLAN.HK.2.5.0.1-01100-QCAHKSWPL_SILICONZ-1
[  139.533358] ath11k c000000.wifi: joba-1 patch V2: ATH11K_QUEUE_LEN=2048
[  143.377821] zram0: detected capacity change from 0 to 212860928
[  143.526674] Adding 207868k swap on /dev/zram0.  Priority:-2 extents:1 across:207868k SSDsc
[  144.785372] ECM init
[  144.785429] ECM database jhash random seed: 0x2fcaaf4
[  144.787732] ECM init complete
[  145.721935] nss-dp 3a003000.dp5 eth0: nss_dp_edma: Registering netdev eth0(qcom-id:5) with EDMA
[  145.928019] br-lan: port 1(eth0) entered blocking state
[  145.928053] br-lan: port 1(eth0) entered disabled state
[  145.932447] device eth0 entered promiscuous mode
[  145.943691] nss-dp 3a007000.dp6 eth1: nss_dp_edma: Registering netdev eth1(qcom-id:6) with EDMA
[  149.214841] nss-dp 3a007000.dp6 eth1: PHY Link up speed: 1000
[  149.214901] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  150.014843] nss-dp 3a003000.dp5 eth0: PHY Link up speed: 1000
[  150.014912] br-lan: port 1(eth0) entered blocking state
[  150.019642] br-lan: port 1(eth0) entered forwarding state
[  150.024919] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  154.583489] pppoe-wan: renamed from ppp0
[  199.650795] ath11k c000000.wifi: failed to load caldata.bin: -110
[  199.650849] ath11k c000000.wifi: qmi failed to load board data file:-110

coolsnowwolf, 2021-11-17:
空间满了

lgh200, 2021-11-17:
> 空间满了

要怎么处理，我固件也不大，才23M左右

closed 2021-11-30 by AmadeusGhost