coolsnowwolf/lede #8271: [教程] 每次更新固件uu加速器总要重新解绑？

greennyreborn, 2021-11-17:
## 解决办法

系统 -> 备份/升级
在配置标签页的内容中添加一行
```/.uuplugin_uuid```

该文件是每次绑定UU加速器后的一个身份标识，更新固件这个位置的文件会被清除，所以导致每次需要重新解绑再绑定。
在备份配置里加入该行后，更新固件这个文件也会保留。
本人测试成功，欢迎大家测试

WYC-2020, 2021-11-18:
知道源码在哪里吗？直接想修改他保存这个文件的位置，这样就能让备份的时候自动备份下来

NagaseKouichi, 2021-11-18:
原来UU在根目录了保存了这个……

源码当然在网易，不会开源的

greennyreborn, 2021-11-19:
> 原来UU在根目录了保存了这个……
> 
> 源码当然在网易，不会开源的

我也是意外发现的

NeedforGit, 2021-11-23:
> ## 解决办法
> 系统 -> 备份/升级 在配置标签页的内容中添加一行 `/.uuplugin_uuid`
> 
> 该文件是每次绑定UU加速器后的一个身份标识，更新固件这个位置的文件会被清除，所以导致每次需要重新解绑再绑定。 在备份配置里加入该行后，更新固件这个文件也会保留。 本人测试成功，欢迎大家测试

感谢分享！ 好人一生平安！

VergilGao, 2021-11-29:
现在uu的openwrt插件最大的问题是只支持主机加速，明明小米华硕的官方插件都已经能支持PC和手机加速了。

VergilGao, 2021-11-30:
这个commit貌似解决了这个问题 https://github.com/coolsnowwolf/lede/commit/525d69c5d7f9553a8ca7faeecc25c5c6c4236f5e

greennyreborn, 2021-11-30:
> 这个commit貌似解决了这个问题 [525d69c](https://github.com/coolsnowwolf/lede/commit/525d69c5d7f9553a8ca7faeecc25c5c6c4236f5e)

是的，大佬们太厉害了

closed 2021-11-30 by greennyreborn