coolsnowwolf/lede #8278: 编译报错：target_install Error 2

sxzq46, 2021-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译到最后报错，但是能成功生成bin文件

### (2) 路由器型号和固件版本

A：红米ac2100

### (3) 详细日志

`make[6]: Leaving directory '/data/lede/target/linux/ramips/image/lzma-loader'
mv "/data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin.bin" "/data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin"
rm -rf /data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin.src
mkimage -A mips -O linux -T kernel -C none -a 0x80001000 -e 0x80001000 -n 'MIPS OpenWrt Linux-5.4.158'   -d /data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin /data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin.new
Image Name:   MIPS OpenWrt Linux-5.4.158
Created:      Thu Nov 18 09:26:36 2021
Image Type:   MIPS Linux Kernel Image (uncompressed)
Data Size:    15854028 Bytes = 15482.45 KiB = 15.12 MiB
Load Address: 80001000
Entry Point:  80001000
mv /data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin.new /data/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-ramips-mt7621-xiaomi_mi-router-ac2100-initramfs-kernel.bin
make[5]: Leaving directory '/data/lede/target/linux/ramips/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/data/lede/target/linux/ramips'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/data/lede/target/linux'
time: target/linux/install#61.84#15.80#57.37
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/data/lede'
make[1]: *** [target/Makefile:19: /data/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/data/lede'
make: *** [/data/lede/include/toplevel.mk:230: world] Error 2
`

WYC-2020, 2021-11-19:
万能打发，distclean

sxzq46, 2021-11-20:
> 

确实有效...这是仓库有编译残留吗？

closed 2021-11-20 by sxzq46