coolsnowwolf/lede #8279: K3的屏幕在 commit:b442cc5 #8207 后无法正常工作

Crazy365, 2021-11-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

K3的屏幕在 commit:b442cc5 #8207 后无法正常工作，屏幕卡在启动界面保持常亮。经测试，前一 commit:937f01b 的代码编译的固件屏幕正常。请各位大佬看看能否解决，万分感谢🙏

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：斐讯K3

### (3) 详细日志

A：相关日志仅发现一条：
Fri Nov 19 19:02:14 2021 user.err K3Screen[5174]: could not get attrs from serial fd: I/O error

EnnawYang, 2021-11-19:
🚮

xiangfeidexiaohuo, 2021-11-20:
反正很神奇，我也遇到了

yangxu52, 2021-11-22:
Before commit:b442cc5,phicomm k3 'led screen dont work.
自从commit:b442cc5提交以后，K3的屏幕无法正常工作。
Manual running k3screenctrl in shell,returen error: segmentation fault.
手动在shell中运行k3screenctrl会报错误：segmentation fault 。
By reading k3screenctrl source code,i guess this program may unable to operate serial port to communit with led controller.
通过读k3screenctrl源码，猜测可能是因为程序没法正常操作串口与led控制器通讯。
Maybe cause by kernel update some serial port operation.
可能是因为内核更新了串口相关操作导致。
There is core dump file. Please,help locate issue.这是core dump出的文件,请帮忙定位。
[k3screenctrl core dump](https://github.com/yangxu52/Phicomm-K2P-K3-OpenWrt-Firmware/blob/main/k3screenctrl.1637515285.9226.11.core)

yangxu52, 2021-11-24:
There is kernel log before commit:b442cc5 (Part of it，aka oldlog):
这是commit:b442cc5提交之前的内核日志（部分，称旧日志）：
[    0.024214] bounce: pool size: 64 pages
[    0.026754] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    0.027276] printk: console [ttyS0] disabled
[    0.027339] 18000300.serial: ttyS0 at MMIO 0x18000300 (irq = 18, base_baud = 7812500) is a 16550
[    0.497272] printk: console [ttyS0] enabled
[    0.501918] 18000400.serial: ttyS1 at MMIO 0x18000400 (irq = 18, base_baud = 7812500) is a 16550
[    0.511315] bcm2835-rng 18004000.rng: hwrng registered
[    0.517019] nand: device found, Manufacturer ID: 0xec, Chip ID: 0xf1

There is kernel log after commit:b442cc5 (Part of it，aka newlog):
这是commit:b442cc5提之后的内核日志（部分，称新日志）：
[    0.024067] bounce: pool size: 64 pages
[    0.026108] ns-pinmux 1800c100.cru:pinctrl: invalid resource
[    0.026123] ns-pinmux 1800c100.cru:pinctrl: Failed to map pinctrl regs
[    0.026140] ns-pinmux: probe of 1800c100.cru:pinctrl failed with error -22
[    0.026513] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
[    0.027029] printk: console [ttyS0] disabled
[    0.027089] 18000300.serial: ttyS0 at MMIO 0x18000300 (irq = 18, base_baud = 7812500) is a 16550
[    0.516090] printk: console [ttyS0] enabled
[    0.520967] bcm2835-rng 18004000.rng: hwrng registered
[    0.526675] nand: device found, Manufacturer ID: 0xec, Chip ID: 0xf1

By comparison，the newlog miss a line:18000400.serial: ttyS1 at MMIO 0x18000400 (irq = 18, base_baud = 7812500) is a 16550.
通过对比，新日志丢失了一行：18000400.serial: ttyS1 at MMIO 0x18000400 (irq = 18, base_baud = 7812500) is a 16550。
But in the fifth line of new log “Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled”，print "2 port" ,as same as oldlog.
但是在新日志的第五行，打印了2端口，和旧的日志一样。
Maybe it was this problem,ttyS1 not init by kernel,that caused k3screenctrl cant opeart serial port(setup failure)
也许正是ttyS1内核未初始化，导致了k3screenctrl无法正常操作串口。

zwillhill, 2021-11-28:
一样的问题，暂时只能把屏幕线拔了。

xiangfeidexiaohuo, 2021-12-02:
估计没有大佬修了，退回养老吧

dgyangxf, 2021-12-09:
> 

可以clone #b442cc5之前的源码编译，git checkout 937f01bea2e216ac5719d0d30b3db6142f920cf3

yangxu52, 2021-12-11:
> > 
> 
> 可以clone #b442cc5之前的源码编译，git checkout [937f01b](https://github.com/coolsnowwolf/lede/commit/937f01bea2e216ac5719d0d30b3db6142f920cf3)

我修改kernel-verison.mk选择5.4.150内核，并且使用该内核提交的源代码中的target文件夹替换掉当前的target文件夹以保证patches相同，并且成功编译出了5.4.150内核的固件但是k3screenctrl依然不能正常工作，依然是segmentaion fault

dgyangxf, 2021-12-12:
哦，那可能是你别的地方没弄好吧，我自己编译的可以用，你可以换我的方法试一下。

yangxu52, 2021-12-12:
> 

我知道你说的checkout切换到以前的提交，我只想用新源码+旧内核

xiangfeidexiaohuo, 2021-12-18:
没人修咯，没人管K3了

barry-ran, 2021-12-29:
期待修复啊

zhangxiaoliangzxl, 2021-12-29:
我已经修复了，在恩山发了帖子，但有人说没有问题，我又删了

zwillhill, 2021-12-29:
> 

看了下那人说的应该不是一回事。5.4.158内核开始才出来的bug，哪有好几年。

ZhangCharlie, 2021-12-29:
我在恩山重新上传了固件，如需要代码补丁，可以@我，几年前我搞的天气屏幕代码一开始没开放就是一些碎嘴子烦的很，后来还是放出来给大家用了

barry-ran, 2021-12-30:
@ZhangCharlie 可以提个pr到lede吗

ZhangCharlie, 2021-12-30:
因为修改了dts, 有可能会影响到其它board, 就不提pr了, 有需要的自己本地打补丁编译吧
https://github.com/ZhangCharlie/k3screen-fix-patch.git

GTXLEDE, 2021-12-31:
> 因为修改了dts, 有可能会影响到其它board, 就不提pr了, 有需要的自己本地打补丁编译吧 https://github.com/ZhangCharlie/k3screen-fix-patch.git

@ZhangCharlie 修改了dts编译出现问题这个需要怎么更改

ZhangCharlie, 2021-12-31:
> 

编译出现什么问题了，你把错误贴上来，而且这个内核补丁是修复5.4.15x版本后没有启用ttyS1的问题，如果原来没有这个问题的就不需要这个补丁

GTXLEDE, 2021-12-31:
> > 
> 
> 编译出现什么问题了，你把错误贴上来，而且这个内核补丁是修复5.4.15x版本后没有启用ttyS1的问题，如果原来没有这个问题的就不需要这个补丁

@ZhangCharlie 
Run echo "Compile_Date=$(date +%Y%m%d%H%M)" >> $GITHUB_ENV
Cloning into 'package/luci-app-passwall'...
Cloning into 'package/luci-app-ssr-plus'...
Cloning into '/home/runner/work/LEDE-k3/LEDE-k3/openwrt/openwrt-package'...
sh -c 'cat '\''/home/runner/work/LEDE-k3/LEDE-k3/openwrt/build/Lede_source/patches/000-k3screen.patch'\''  | patch -d '\''./'\'' -p1 --forward --no-backup-if-mismatch' 
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
sh -c 'cat '\''/home/runner/work/LEDE-k3/LEDE-k3/openwrt/build/Lede_source/patches/906-BCM5301x-uart1.patch'\''  | patch -d '\''./'\'' -p1 --forward --no-backup-if-mismatch' 
The text leading up to this was:
--------------------------
|diff -Nuar a/src/infocenter.c b/src/infocenter.c
|--- a/src/infocenter.c	2017-05-02 19:49:08.000000000 +0800
|+++ b/src/infocenter.c	2021-12-29 17:37:00.837529076 +0800
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
2 out of 2 hunks ignored
can't find file to patch at input line 4
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff -Nuar a/arch/arm/boot/dts/bcm5301x.dtsi b/arch/arm/boot/dts/bcm5301x.dtsi
|--- a/arch/arm/boot/dts/bcm5301x.dtsi	2021-12-29 16:55:29.441523015 +0800
|+++ b/arch/arm/boot/dts/bcm5301x.dtsi	2021-12-29 17:03:44.640694605 +0800
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
Error: Process completed with exit code 123.

ZhangCharlie, 2021-12-31:
> ## @ZhangCharlie
> Run echo "Compile_Date=$(date +%Y%m%d%H%M)" >> $GITHUB_ENV
> Cloning into 'package/luci-app-passwall'...
> Cloning into 'package/luci-app-ssr-plus'...
> Cloning into [...]

看来小白还不少啊，这个是openwrt的patch, 放在下面位置直接编译就好了, k3screenctrl的位置根据自己的调整，大多数人都没有用lean原来的k3screenctrl

xxx/target/linux/bcm53xx/patches-5.4/906-BCM5301x-uart1.patch
xxx/package/lean/k3screenctrl/patches/000-k3screen.patch

GTXLEDE, 2021-12-31:
> 

@ZhangCharlie 我是直接在线编译的，不是本地编译，https://github.com/GTXLEDE/LEDE-k3 补丁可以直接导入，路径设置过还是一样的问题

ZhangCharlie, 2022-01-01:
@GTXLEDE 可能和你自己的代码有冲突，可以自己根据patch里的内容，在自己的代码上修改后打patch, 我本地是没问题的

Applying /home/zxl/work/lean/target/linux/bcm53xx/patches-5.4/800-0002-nvmem-brcm_nvram-provide-NVMEM-content-to-the-NVRAM-.patch using plaintext: 
patching file drivers/nvmem/brcm_nvram.c

Applying /home/zxl/work/lean/target/linux/bcm53xx/patches-5.4/905-BCM53573-minor-hacks.patch using plaintext: 
patching file arch/arm/boot/dts/bcm53573.dtsi
patching file drivers/bcma/main.c
patching file drivers/clocksource/arm_arch_timer.c

Applying /home/zxl/work/lean/target/linux/bcm53xx/patches-5.4/906-BCM5301x-uart1.patch using plaintext: 
patching file arch/arm/boot/dts/bcm5301x.dtsi
touch /home/zxl/work/lean/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.162/.prepared_4f7ee7c9a1d44d055fcdc638c2251e17
make[3]: Leaving directory '/home/zxl/work/lean/target/linux/bcm53xx'
make[2]: Leaving directory '/home/zxl/work/lean/target/linux'

yangxu52, 2022-01-01:
> 因为修改了dts, 有可能会影响到其它board, 就不提pr了, 有需要的自己本地打补丁编译吧 https://github.com/ZhangCharlie/k3screen-fix-patch.git

我之前研究也注意到了ttyS1的问题，但是我使用新的源码，更改include/kernerl-version.mk的内核版本呢，并且用旧的target目录替换（为了旧的内核对应的patch），然后成功编译出来旧的内存的固件（5.4.150 152 156我都试了），ttyS1也在内核日志里初始化了，但是屏幕还是挂的。
所以如果只是ttyS1导致的话，那确实很玄学。我对Linux编程不熟悉，研究没法深入，还有就是你看上面我节选的内核日志的对比，新内核还多了几行ns-pinmux的日志，是否有关联。

我现在正在编译打了你的patch的固件明天试试效果。

Crazy365, 2022-01-02:
感谢 @ZhangCharlie 大佬！感谢大雕 @coolsnowwolf ！终于可以继续升级了！

closed 2022-01-02 by Crazy365