coolsnowwolf/lede #8283: Action编译出现 naiveproxy failed to build

qqsir-dev, 2021-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
ERROR: package/lean/helloworld/naiveproxy failed to build
A：

### (2) 路由器型号和固件版本

A：软路由

### (3) 详细日志

A：

Dimitri2020007, 2021-11-20:
naiveproxy又更新版本了？

rq1025330, 2021-11-20:
也是

binge8, 2021-11-20:
chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
这个下载失败

haiibo, 2021-11-20:
同上

WYC-2020, 2021-11-20:
我action一切正常

Dimitri2020007, 2021-11-20:
我action编译node报错了

rq1025330, 2021-11-20:
> 我action编译node报错了，naiveproxy也报错了y

> 我action编译node报错了，naiveproxy也报错了

一样

ostrichii, 2021-11-20:
一样
日志：
SHELL= flock /home/ostrich/r4s/lede/tmp/.expat-2.4.1.tar.xz.flock -c '          /home/ostrich/r4s/lede/scripts/download.pl "/home/ostrich/r4s/lede/dl" "expat-2.4.1.tar.xz" "cf032d0dba9b928636548e32b327a2d66b1aab63c4f4a13dd132c2d1d2f2fb6a" "" "@SF/expat"    '
make[3]: Leaving directory '/home/ostrich/r4s/lede/feeds/packages/libs/expat'
time: package/feeds/packages/expat/host-compile#0.16#0.04#0.20
make[3]: Entering directory '/home/ostrich/r4s/lede/feeds/packages/libs/libffi'
mkdir -p /home/ostrich/r4s/lede/dl
SHELL= flock /home/ostrich/r4s/lede/tmp/.libffi-3.3.tar.gz.flock -c '   /home/ostrich/r4s/lede/scripts/download.pl "/home/ostrich/r4s/lede/dl" "libffi-3.3.tar.gz" "72fba7922703ddfa7a028d513ac15a85c8d54c8d67f55fa5a4802885dc652056" "" "https://sourceware.org/pub/libffi/"    '
make[3]: Leaving directory '/home/ostrich/r4s/lede/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.18#0.04#0.22
make[3]: Entering directory '/home/ostrich/r4s/lede/package/utils/bzip2'
mkdir -p /home/ostrich/r4s/lede/dl
SHELL= flock /home/ostrich/r4s/lede/tmp/.bzip2-1.0.8.tar.gz.flock -c '          /home/ostrich/r4s/lede/scripts/download.pl "/home/ostrich/r4s/lede/dl" "bzip2-1.0.8.tar.gz" "ab5a03176ee106d3f0fa90e381da478ddae405918153cca248e682cd0c4a2269" "" "https://sourceware.org/pub/bzip2"    '
make[3]: Leaving directory '/home/ostrich/r4s/lede/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.17#0.02#0.18
make[3]: Entering directory '/home/ostrich/r4s/lede/feeds/packages/lang/python/python3'
mkdir -p /home/ostrich/r4s/lede/dl
SHELL= flock /home/ostrich/r4s/lede/tmp/.Python-3.9.5.tar.xz.flock -c '         /home/ostrich/r4s/lede/scripts/download.pl "/home/ostrich/r4s/lede/dl" "Python-3.9.5.tar.xz" "0c5a140665436ec3dbfbb79e2dfb6d192655f26ef4a29aeffcb6d1820d716d83" "" "https://www.python.org/ftp/python/3.9.5"    '
make[3]: Leaving directory '/home/ostrich/r4s/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.32#0.07#0.36
make[3]: Entering directory '/home/ostrich/r4s/lede/feeds/helloworld/naiveproxy'
mkdir -p /home/ostrich/r4s/lede/dl
SHELL= flock /home/ostrich/r4s/lede/tmp/.chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata.flock -c '      /home/ostrich/r4s/lede/scripts/download.pl "/home/ostrich/r4s/lede/dl" "chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata" "8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f" "chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata" "https://storage.googleapis.com.cnpmjs.org/chromium-optimization-profiles/pgo_profiles" "https://storage.googleapis.com/chromium-optimization-profiles/pgo_profiles"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://storage.googleapis.com.cnpmjs.org/chromium-optimization-profiles/pgo_profiles/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 47.5M  100 47.5M    0     0  3508k      0  0:00:13  0:00:13 --:--:-- 3635k
Hash of the downloaded file does not match (file: 5f693c4779c1d6802e509c2c1e5f130017fd6634a0ccd3a0de47be419c068f5e, requested: 8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://storage.googleapis.com/chromium-optimization-profiles/pgo_profiles/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 47.5M  100 47.5M    0     0  3322k      0  0:00:14  0:00:14 --:--:-- 3407k
Hash of the downloaded file does not match (file: 5f693c4779c1d6802e509c2c1e5f130017fd6634a0ccd3a0de47be419c068f5e, requested: 8dcf5973033d40c9a7b15e571dea3832e7b67976aad9113369e22d43808c603f) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/chrome-linux-4664-1636557077-6e390f4e505916531ca2ab0c895d5903ab4d88a9.profdata
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
[... cut ...]
package/Makefile:114: recipe for target 'package/feeds/helloworld/naiveproxy/compile' failed
make[2]: *** [package/feeds/helloworld/naiveproxy/compile] Error 1
make[2]: Leaving directory '/home/ostrich/r4s/lede'
package/Makefile:110: recipe for target '/home/ostrich/r4s/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/ostrich/r4s/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ostrich/r4s/lede'
/home/ostrich/r4s/lede/include/toplevel.mk:228: recipe for target 'world' failed

xylz0928, 2021-11-20:
https://github.com/fw876/helloworld/pull/721

haoranxu1989, 2021-11-20:
一样的报错，从昨天开始的，之前不报错

ostrichii, 2021-11-20:
测试好了，更新代码再试试

1054009064, 2021-11-21:
https://github.com/fw876/helloworld/commit/545f5405b57b72aaaace0c255eb53e1d1274f55d

closed 2021-11-21 by qqsir-dev