coolsnowwolf/lede #8284: FTP error是什么原因

XuYuQuan0713, 2021-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
在执行make download -j8 V=s后出现FTP 错误，过程中挂了代理执行的下载，在阿里云服务器上
### (1) 具体问题

make[2]: Leaving directory '/home/juzhen/lede/tools/zlib'
time: tools/zlib/download#0.10#0.04#9.76
  0  613k    0     0    0     0      0      0 --:--:--  0:01:03 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:38355  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:57496  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:10108  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:51641  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:00:38 --:--:--     0

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：make[2]: Leaving directory '/home/juzhen/lede/tools/zlib'
time: tools/zlib/download#0.10#0.04#9.76
  0  613k    0     0    0     0      0      0 --:--:--  0:01:03 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:38355  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:57496  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:10108  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:51641  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 16 seconds. 1 retries 
Warning: left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:14822  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:00:49 --:--:--     0
make[2]: Leaving directory '/home/juzhen/lede/tools/zlib'
time: tools/zlib/download#0.10#0.04#9.76
  0  613k    0     0    0     0      0      0 --:--:--  0:01:03 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:38355  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:57496  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:10108  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:51641  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:01:00 --:--:--     0
curl: (19) RETR response: 425
Warning: Transient problem: FTP error Will retry in 16 seconds. 1 retries 
Warning: left.
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0[proxychains] Strict chain  ...  127.0.0.1:1080  ...  127.0.0.1:14822  ...  OK
  0  613k    0     0    0     0      0      0 --:--:--  0:00:49 --:--:--     0

WYC-2020, 2021-11-20:
就下载失败还能有啥原因，编译的前提条件梯子，要么就是梯子不行要么就是资源没有了

closed 2021-11-21 by coolsnowwolf

XuYuQuan0713, 2021-11-21:
解决了，不影响编译