coolsnowwolf/lede #8286: Luci界面无法配置IPv6

ostrichii, 2021-11-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
WAN口和LAN口的IPv6设置失效，默认WAN口无法禁用IPv6，LAN口无法启用。16号代码是正常的
### (1) 具体问题
[image]
该配置设置无效果
[image]

A：

### (2) 路由器型号和固件版本
R4S，OpenWrt R21.11.11最新版本代码
A：

### (3) 详细日志

A：
Sun Nov 21 07:22:55 2021 daemon.info pppd[2951]: Terminating on signal 15
Sun Nov 21 07:22:55 2021 daemon.info pppd[2951]: Connect time 1.0 minutes.
Sun Nov 21 07:22:55 2021 daemon.info pppd[2951]: Sent 368260 bytes, received 617173 bytes.
Sun Nov 21 07:22:55 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sun Nov 21 07:22:55 2021 daemon.notice pppd[2951]: Connection terminated.
Sun Nov 21 07:22:55 2021 daemon.info pppd[2951]: Connect time 1.0 minutes.
Sun Nov 21 07:22:55 2021 daemon.info pppd[2951]: Sent 368260 bytes, received 617173 bytes.
Sun Nov 21 07:22:56 2021 daemon.info pppd[2951]: Exit.
Sun Nov 21 07:22:56 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 21 07:22:56 2021 kern.info kernel: [  100.736559] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
Sun Nov 21 07:22:56 2021 daemon.notice netifd: Interface 'wan' is disabled
Sun Nov 21 07:22:56 2021 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sun Nov 21 07:22:56 2021 kern.info kernel: [  100.805696] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=85)
Sun Nov 21 07:22:56 2021 kern.err kernel: [  100.807348] cma: cma_alloc: alloc failed, req-size: 2 pages, ret: -12
Sun Nov 21 07:22:56 2021 kern.err kernel: [  100.807975] cma: cma_alloc: alloc failed, req-size: 2 pages, ret: -12
Sun Nov 21 07:22:56 2021 daemon.notice netifd: Interface 'wan' is enabled
Sun Nov 21 07:22:56 2021 kern.info kernel: [  100.819722] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
Sun Nov 21 07:22:56 2021 kern.warn kernel: [  100.820421] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
Sun Nov 21 07:22:56 2021 kern.info kernel: [  100.821046] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
Sun Nov 21 07:22:56 2021 user.notice mwan3[6283]: Execute ifdown event on interface wan (unknown)
Sun Nov 21 07:22:56 2021 user.info mwan3track[3867]: Detect ifdown event on interface wan (pppoe-wan)
Sun Nov 21 07:22:56 2021 user.info mwan3track[3867]: Check (ping) failed for target "xxx.xxx.xxx.xxx" on interface wan (pppoe-wan)
Sun Nov 21 07:22:59 2021 daemon.notice netifd: Network device 'eth0' link is up
Sun Nov 21 07:22:59 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Nov 21 07:22:59 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 21 07:22:59 2021 kern.info kernel: [  103.914312] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Sun Nov 21 07:22:59 2021 kern.info kernel: [  103.915091] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sun Nov 21 07:22:59 2021 daemon.err insmod: module is already loaded - slhc
Sun Nov 21 07:22:59 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 21 07:22:59 2021 daemon.err insmod: module is already loaded - pppox
Sun Nov 21 07:22:59 2021 daemon.err insmod: module is already loaded - pppoe
Sun Nov 21 07:22:59 2021 daemon.notice netifd: wan (7060): uci: Entry not found
Sun Nov 21 07:22:59 2021 daemon.notice netifd: wan (7060): sh: out of range
Sun Nov 21 07:22:59 2021 daemon.info pppd[7070]: Plugin pppoe.so loaded.
Sun Nov 21 07:22:59 2021 daemon.info pppd[7070]: PPPoE plugin from pppd 2.4.9
Sun Nov 21 07:22:59 2021 daemon.notice pppd[7070]: pppd 2.4.9 started by root, uid 0
Sun Nov 21 07:23:01 2021 user.notice mwan3track[3867]: Interface wan (pppoe-wan) is offline
Sun Nov 21 07:23:04 2021 daemon.info pppd[7070]: PPP session is 29271
Sun Nov 21 07:23:04 2021 daemon.warn pppd[7070]: Connected to e0:24:7f:f9:af:14 via interface eth0
Sun Nov 21 07:23:04 2021 kern.info kernel: [  109.350892] pppoe-wan: renamed from ppp0
Sun Nov 21 07:23:04 2021 daemon.info pppd[7070]: Renamed interface ppp0 to pppoe-wan
Sun Nov 21 07:23:04 2021 daemon.info pppd[7070]: Using interface pppoe-wan
Sun Nov 21 07:23:04 2021 daemon.notice pppd[7070]: Connect: pppoe-wan <--> eth0
Sun Nov 21 07:23:07 2021 daemon.info pppd[7070]: syncppp not active
Sun Nov 21 07:23:07 2021 daemon.info pppd[7070]: Remote message: Authentication success,Welcome!
Sun Nov 21 07:23:07 2021 daemon.notice pppd[7070]: PAP authentication succeeded
Sun Nov 21 07:23:07 2021 daemon.notice pppd[7070]: peer from calling number E0:24:7F:F9:AF:14 authorized
[... 61 lines cut ...]
Sun Nov 21 07:23:53 2021 daemon.info pppd[9180]: PPP session is 12600
Sun Nov 21 07:23:53 2021 daemon.warn pppd[9180]: Connected to e0:24:7f:f9:af:14 via interface eth0
Sun Nov 21 07:23:53 2021 kern.info kernel: [  158.372896] pppoe-wan: renamed from ppp0
Sun Nov 21 07:23:53 2021 daemon.info pppd[9180]: Renamed interface ppp0 to pppoe-wan
Sun Nov 21 07:23:53 2021 daemon.info pppd[9180]: Using interface pppoe-wan
Sun Nov 21 07:23:53 2021 daemon.notice pppd[9180]: Connect: pppoe-wan <--> eth0
Sun Nov 21 07:23:56 2021 daemon.info pppd[9180]: syncppp not active
Sun Nov 21 07:23:56 2021 daemon.info pppd[9180]: Remote message: Authentication success,Welcome!
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: PAP authentication succeeded
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: peer from calling number E0:24:7F:F9:AF:14 authorized
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: local  IP address xxx.xxx.xxx.xxx
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: remote IP address xxx.xxx.xxx.xxx
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: primary   DNS address xxx.xxx.xxx.xxx
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: secondary DNS address xxx.xxx.xxx.xxx
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: local  LL address fe80::9540:1603:a3ee:f807
Sun Nov 21 07:23:56 2021 daemon.notice pppd[9180]: remote LL address fe80::e224:7fff:fef9:af14
Sun Nov 21 07:23:56 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Nov 21 07:23:56 2021 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sun Nov 21 07:23:56 2021 daemon.notice netifd: Interface 'wan' is now up
Sun Nov 21 07:23:57 2021 user.notice mwan3[9343]: Execute ifup event on interface wan (pppoe-wan)
Sun Nov 21 07:23:57 2021 user.notice mwan3[9343]: Starting tracker on interface wan (pppoe-wan)
Sun Nov 21 07:23:57 2021 user.notice mwan3track[7608]: Stopping mwan3track for interface "wan"
Sun Nov 21 07:23:58 2021 user.info mwan3rtmon[2159]: Detect rtchange event.
Sun Nov 21 07:23:58 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sun Nov 21 07:23:59 2021 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)

xiaobaishu1, 2021-11-21:
你都没按IPV6的依赖

MGBB, 2021-11-21:
> 你没有按IPV6的依赖

他的意思是在高级设置里关闭了获取ipv6地址选项，在接口总览页面还是会有ipv6地址，确实有这样问题我也遇到了

ostrichii, 2021-11-21:
> 你都没按IPV6的依赖

不是，编译配置没改过，16号的代码编译没问题。

ostrichii, 2021-11-21:
> > 你没有按IPV6的依赖
> 
> 他的意思是在高级设置里选择了禁用ipv6地址，在接口总览页面还是会有ipv6地址，即使编译时不选择ipv6依赖也会会出现在这种问题，这样问题我也遇到了

是的

suizhenmin, 2021-11-22:
求教，iptv如何进行设置

WYC-2020, 2021-11-22:
iptv 这种你怕是跑错地方了，自己去ytb

ostrichii, 2021-11-22:
> iptv 这种你怕是跑错地方了，自己去ytb

IPTV和宽带内网融合，我跑了两年了，一直没啥问题

ostrichii, 2021-11-22:
> 求教，iptv如何进行设置

各地的IPTV配置鉴权有区别，当地的鉴权方式可以恩山找找看有没有。

suizhenmin, 2021-11-22:
> > 求教，iptv如何进行设置
> 
> 哈里的IPTV配置鉴权有区别，当地的鉴权方式可以恩山找找看有没有。

有详细的教程么，一直没找到

jliveusa, 2021-11-22:
同样遇到这个问题，还有frp的luci也受影响了

ja5onwu, 2021-11-23:
我编译的也有这个问题,关了IPV6后,依然有IPV6的地址

suizhenmin, 2021-11-23:
我昨天编译的，在概览页面没有，你到网络接口那是可以看到
[image]

NeedforGit, 2021-11-23:
跟楼主一样，也是这个问题，wan口禁用了ipv6, 但还一直显示ipv6, 希望能修复！

pipihu, 2021-11-23:
哈哈哈哈，这应该是升级ppp 2.4.9 造成的。
修改 https://github.com/coolsnowwolf/lede/blob/master/package/network/services/ppp/files/ppp.sh 文件
（参考官方的 https://github.com/openwrt/openwrt/blob/master/package/network/services/ppp/files/ppp.sh 文件）
第152行
```
		ipv6-up-script /lib/netifd/ppp6-up \
```
改为
```
		${ipv6:+ipv6-up-script /lib/netifd/ppp6-up} \
```
第154行
```
		ipv6-down-script /lib/netifd/ppp-down \
```
改为
```
		${ipv6:+ipv6-down-script /lib/netifd/ppp-down} \
```
关闭ipv6之后接口不再显示ipv6地址
[image]
```
ubus call network.interface.ADSL status
```
[image]

@coolsnowwolf
等大佬修复

closed 2021-11-24 by AmadeusGhost