coolsnowwolf/lede #8292: 内核版本:5.10.80 拨不了号

lihonfu, 2021-11-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
WAN显示是灰色的
### (1) 具体问题

A：PPPOE拨不了号

### (2) 路由器型号和固件版本

A：路由型号:Linksys1900acsv2
内核版本:内核版本:5.10.80
Powered by LuCI Master (git-21.320.49600-21168d4) / OpenWrt R21.11.11

### (3) 详细日志

A：Sun Nov 21 14:52:49 2021 daemon.info pppd[504]: Exit.
Sun Nov 21 14:52:49 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 21 14:52:49 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - slhc
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - pppox
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - pppoe
Sun Nov 21 14:52:49 2021 daemon.notice netifd: wan (531): uci: Entry not found
Sun Nov 21 14:52:49 2021 daemon.notice netifd: wan (531): sh: out of range
Sun Nov 21 14:52:49 2021 daemon.info pppd[564]: Plugin pppoe.so loaded.
Sun Nov 21 14:52:49 2021 daemon.info pppd[564]: PPPoE plugin from pppd 2.4.9
Sun Nov 21 14:52:49 2021 daemon.notice pppd[564]: pppd 2.4.9 started by root, uid 0
Sun Nov 21 14:52:49 2021 daemon.warn pppd[564]: Timeout waiting for PADO packets
Sun Nov 21 14:52:49 2021 daemon.err pppd[564]: Unable to complete PPPoE Discovery
Sun Nov 21 14:52:49 2021 daemon.info pppd[564]: Exit.
Sun Nov 21 14:52:49 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 21 14:52:49 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - slhc
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - pppox
Sun Nov 21 14:52:49 2021 daemon.err insmod: module is already loaded - pppoe
Sun Nov 21 14:52:49 2021 daemon.notice netifd: wan (591): uci: Entry not found
Sun Nov 21 14:52:49 2021 daemon.notice netifd: wan (591): sh: out of range
Sun Nov 21 14:52:49 2021 daemon.info pppd[624]: Plugin pppoe.so loaded.
Sun Nov 21 14:52:49 2021 daemon.info pppd[624]: PPPoE plugin from pppd 2.4.9
Sun Nov 21 14:52:49 2021 daemon.notice pppd[624]: pppd 2.4.9 started by root, uid 0
Sun Nov 21 14:52:49 2021 daemon.warn pppd[624]: Timeout waiting for PADO packets
Sun Nov 21 14:52:49 2021 daemon.err pppd[624]: Unable to complete PPPoE Discovery
Sun Nov 21 14:52:49 2021 daemon.info pppd[624]: Exit.
Sun Nov 21 14:53:04 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 21 14:53:04 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - slhc
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - pppox
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - pppoe
Sun Nov 21 14:53:04 2021 daemon.notice netifd: wan (4764): uci: Entry not found
Sun Nov 21 14:53:04 2021 daemon.notice netifd: wan (4764): sh: out of range
Sun Nov 21 14:53:04 2021 daemon.info pppd[4797]: Plugin pppoe.so loaded.
Sun Nov 21 14:53:04 2021 daemon.info pppd[4797]: PPPoE plugin from pppd 2.4.9
Sun Nov 21 14:53:04 2021 daemon.notice pppd[4797]: pppd 2.4.9 started by root, uid 0
Sun Nov 21 14:53:04 2021 daemon.warn pppd[4797]: Timeout waiting for PADO packets
Sun Nov 21 14:53:04 2021 daemon.err pppd[4797]: Unable to complete PPPoE Discovery
Sun Nov 21 14:53:04 2021 daemon.info pppd[4797]: Exit.
Sun Nov 21 14:53:04 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 21 14:53:04 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - slhc
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - pppox
Sun Nov 21 14:53:04 2021 daemon.err insmod: module is already loaded - pppoe
Sun Nov 21 14:53:04 2021 daemon.notice netifd: wan (4824): uci: Entry not found
Sun Nov 21 14:53:04 2021 daemon.notice netifd: wan (4824): sh: out of range
Sun Nov 21 14:53:04 2021 daemon.info pppd[4856]: Plugin pppoe.so loaded.
Sun Nov 21 14:53:04 2021 daemon.info pppd[4856]: PPPoE plugin from pppd 2.4.9
Sun Nov 21 14:53:05 2021 daemon.notice pppd[4856]: pppd 2.4.9 started by root, uid 0
Sun Nov 21 14:53:05 2021 daemon.warn pppd[4856]: Timeout waiting for PADO packets
Sun Nov 21 14:53:05 2021 daemon.err pppd[4856]: Unable to complete PPPoE Discovery
Sun Nov 21 14:53:05 2021 daemon.info pppd[4856]: Exit
[image]
[image]
.

lihonfu, 2021-11-22:
固件编译很多次了都是这样子的!求救!

ja5onwu, 2021-11-23:
先用猫拨号,切换去DHCP看看WAN口有没坏先

lihonfu, 2021-11-24:
> 先用猫拨号，切换去DHCP看看WAN口有没坏先

还原上个版本的固件就没问题!WAN口绝对没坏!

lihonfu, 2021-11-24:
> 我的竞斗云也一样，升到5.10版本后，切换到PPPOE协议显示MAC地址，手动指定也不行，DHCP模式下可以正常显示MAC；同的配置，退回到5.4版本就可拨号

情况跟你一模一样!郁闷了..不知道什么情况.兄弟你解决没有?

lihonfu, 2021-11-24:
> 我的竞斗云也一样，升到5.10版本后，切换到PPPOE协议就不显示MAC地址，手动指定也不行，DHCP模式下可以正常显示MAC； 同样的配置，退回到5.4版本就可正常拨号

我把虚拟机还原快照,还原到第一次编译的状态完成后再编译貌似没问题了.

acooler15, 2021-11-26:
我的k2p也是，不清楚啥情况。。。

blueveryday, 2021-11-27:
HC5962，最新的代码，pppoe无法拨号。
和你们的问题一样。
以前的版本号：r3847-4b32d3505可以正常使用。

hzjixiang, 2021-11-28:
今天编译的也没办法拨号，PADO超时

acooler15, 2021-11-28:
我清理了编译缓存，再次编译后成功

blueveryday, 2021-11-29:
> 

还能这样操作啊？

blueveryday, 2021-12-04:
> > > 
> > 
> > 
> > 经过测试，make distclean以后，重新编译，pppoe拨号能够正常使用了。
> 
> 确定吗？ 我把文件夹删掉重新来

真的，你make clean或者make distclean。
但是我发现如果git pull以后，重新编译，拨号又会不行了，我不晓得原因，再make clean，重新编译，重新刷机，不要保留设置，又会好了。
原因不明，反正我是这么搞好的。

hzjixiang, 2021-12-04:
> > > > 
> > > 
> > > 
> > > 经过测试，make distclean以后，重新编译，pppoe拨号能够正常使用了。
> > 
> > 
> > 确定吗？ 我把文件夹删掉重新来
> 
> 真的，你make clean或者make distclean。 但是我发现如果git pull以后，重新编译，拨号又会不行了，我不晓得原因，再make clean，重新编译，重新刷机，不要保留设置，又会好了。 原因不明，反正我是这么搞好的。

谢了。我删除文件夹重新拉取好了

closed 2021-12-07 by AmadeusGhost