coolsnowwolf/lede #8296: WSL2 Ubuntu 20 LTS 编译报错 虚拟机下不会

epenli, 2021-11-23:


epenli, 2021-11-23:
ad@X470:/mnt/c/Users/ad/lede$ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin make -j1 V=s
make[2]: Warning: File '/mnt/c/Users/ad/lede/tmp/info/.files-packageinfo.mk' has modification time 1.1 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File '/mnt/c/Users/ad/lede/tmp/info/.files-targetinfo.mk' has modification time 0.86 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File '/mnt/c/Users/ad/lede/tmp/.packageauxvars' has modification time 0.49 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Entering directory '/mnt/c/Users/ad/lede'
make[2]: Entering directory '/mnt/c/Users/ad/lede'
+ mkdir -p /mnt/c/Users/ad/lede/staging_dir/target-x86_64_musl
+ cd /mnt/c/Users/ad/lede/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /mnt/c/Users/ad/lede/build_dir/target-x86_64_musl/stamp
touch /mnt/c/Users/ad/lede/staging_dir/target-x86_64_musl/.prepared
+ mkdir -p /mnt/c/Users/ad/lede/staging_dir/host
+ cd /mnt/c/Users/ad/lede/staging_dir/host
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /mnt/c/Users/ad/lede/build_dir/host/stamp /mnt/c/Users/ad/lede/staging_dir/host/include/sys
install -m0644 /mnt/c/Users/ad/lede/tools/include/*.h /mnt/c/Users/ad/lede/staging_dir/host/include/
install -m0644 /mnt/c/Users/ad/lede/tools/include/sys/*.h /mnt/c/Users/ad/lede/staging_dir/host/include/sys/
ln -snf lib /mnt/c/Users/ad/lede/staging_dir/host/lib64
touch /mnt/c/Users/ad/lede/staging_dir/host/.prepared
make[3]: Entering directory '/mnt/c/Users/ad/lede/tools/flock'
make[3]: Leaving directory '/mnt/c/Users/ad/lede/tools/flock'
time: tools/flock/compile#0.11#0.04#0.61
make[3]: Entering directory '/mnt/c/Users/ad/lede/tools/xz'
(cd /mnt/c/Users/ad/lede/build_dir/host/xz-5.2.5/; if [ -x configure ]; then cp -fpR /mnt/c/Users/ad/lede/scripts/config.{guess,sub} /mnt/c/Users/ad/lede/build_dir/host/xz-5.2.5// && CC="gcc" CFLAGS="-O2 -I/mnt/c/Users/ad/lede/staging_dir/host/include " CXX="g++" CPPFLAGS="-I/mnt/c/Users/ad/lede/staging_dir/host/include " LDFLAGS="-L/mnt/c/Users/ad/lede/staging_dir/host/lib " CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/mnt/c/Users/ad/lede/staging_dir/host --exec-prefix=/mnt/c/Users/ad/lede/staging_dir/host --sysconfdir=/mnt/c/Users/ad/lede/staging_dir/host/etc --localstatedir=/mnt/c/Users/ad/lede/staging_dir/host/var --sbindir=/mnt/c/Users/ad/lede/staging_dir/host/bin --enable-static=yes --enable-shared=no --disable-doc --disable-nls --with-pic ; fi )
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory
cat: -: No such file or directory

XZ Utils 5.2.5

System type:
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu

Configure options:
checking if debugging code should be compiled... no
checking which encoders to build... lzma1 lzma2 delta x86 powerpc ia64 arm armthumb sparc
checking which decoders to build... lzma1 lzma2 delta x86 powerpc ia64 arm armthumb sparc
checking which match finders to build... hc3 hc4 bt2 bt3 bt4
checking which integrity checks to build... crc32 crc64 sha256
checking if external SHA-256 should be used... no
checking if assembler optimizations should be used... x86_64
checking if small size is preferred over speed... no
checking if threading support is wanted... yes, posix
checking how much RAM to assume if the real amount is unknown... 128 MiB
cat: -: No such file or directory
checking if library symbol versioning should be used... yes
checking if sandboxing should be used... maybe (autodetect)

checking for a shell that conforms to POSIX... /bin/sh

Initializing Automake:
checking for a BSD-compatible install... /mnt/c/Users/ad/lede/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... cat: -: No such file or directory
no
checking whether make supports nested variables... yes
cat: -: No such file or directory
cat: -: No such file or directory
checking whether ln -s works... yes
checking whether make supports the include directive... cat: -: No such file or directory
no
checking for x86_64-pc-linux-gnu-gcc... gcc
cat: -: No such file or directory
checking whether the C compiler works... no
configure: error: in `/mnt/c/Users/ad/lede/build_dir/host/xz-5.2.5':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:37: /mnt/c/Users/ad/lede/build_dir/host/xz-5.2.5/.configured] Error 77
make[3]: Leaving directory '/mnt/c/Users/ad/lede/tools/xz'
time: tools/xz/compile#0.47#0.33#2.97
    ERROR: tools/xz failed to build.
make[2]: *** [tools/Makefile:159: tools/xz/compile] Error 1
make[2]: Leaving directory '/mnt/c/Users/ad/lede'
make[1]: *** [tools/Makefile:155: /mnt/c/Users/ad/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/mnt/c/Users/ad/lede'
make: *** [/mnt/c/Users/ad/lede/include/toplevel.mk:230: world] Error 2

epenli, 2021-11-23:
貌似不能在 win 目录下 只能到wsl 自己的目录下

yangxu52, 2021-11-23:
首先必须在wsl自己目录下进行从git clone开始的一切操作，其次编译时候wsl的环境变量会导致问题，删除windows相关环境变量。

closed 2021-11-24 by AmadeusGhost