coolsnowwolf/lede #8298: luci-app-arpbind 无法绑定，建议修正一下。

iseeyou, 2021-11-23:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ip neigh 显示 192.168.1.2 dev br-lan lladdr xx:xx:xx:xx:xx:xxREACHABLE 而不是 permanent，关机后不久就失效

### (2) 路由器型号和固件版本

A：小米AC2100

### (3) 详细日志

A：无

几个月之前就不行，就改为使用 https://github.com/msylgj/OpenWrt_luci-app/tree/main/luci-app-arpbind 一切正常，好久没编译了，今天更新编译了一下没使用上面的，而是自带的，关机后不久就失效。

Beginner-Go, 2021-11-23:
这个东西好像一直都没用吧

iseeyou, 2021-11-23:
> 

远程唤醒电脑很有用。

Beginner-Go, 2021-11-23:
我对比了一下init.d只发现一处不同

Beginner-Go, 2021-11-23:
[按照我这样改试一试](https://github.com/Beginner-Go/lede/commit/56ad7994a7e2b9cdc0b8606bef046d1fa40e2a54)

iseeyou, 2021-11-24:
> [按照我这样改试一试](https://github.com/Beginner-Go/lede/commit/56ad7994a7e2b9cdc0b8606bef046d1fa40e2a54) 然后重启路由器，测试好给我结果

凌晨修改后编译早上起来也能远程唤醒电脑,这样正常了。
ip neigh
192.168.1.4 dev br-lan lladdr xx:xx:xx:xx:xx:xx PERMANENT

closed 2021-11-24 by AmadeusGhost

ksong008, 2021-11-24:
如果没记错的话，这样修改，唤醒之后，还是会掉的

iseeyou, 2021-11-24:
> 

不会的，除非没电路由器断电。
之前我用几个月都正常。

VergilGao, 2021-11-24:
arp绑定其实在dnsmasq里就有可以设置了。

iseeyou, 2021-11-24:
@VergilGao 有教程吗？想试试.谢谢

VergilGao, 2021-11-30:
> @VergilGao 有教程吗？想试试.谢谢

我理解有问题，dnsmasq里设置的是dhcp的静态地址分配，不是arp绑定。