coolsnowwolf/lede #8304: WatchCat 有没有大佬修一修啊。luci界面的参数无法写入配置文件

82kg, 2021-11-25:
老BUG了

WYC-2020, 2021-11-25:
有吗？等会我看下勒

WYC-2020, 2021-11-25:
配置写在system的

WYC-2020, 2021-11-25:
不知道为啥这样改，后面在细看一下

WYC-2020, 2021-11-25:
@82kg 已修复

WYC-2020, 2021-11-25:
[image]
自己去改

WYC-2020, 2021-11-25:
还有一个判断配置文件是否存在哪里没改，你自己改下吧，其他就没了，就改这三个地方

xyskwjy, 2021-11-25:
> @82kg 已修复

源码修复了吗？还是手动自己修？

WYC-2020, 2021-11-25:
肯定修复源码撒，没有提pr 自己去改 改了提pr @82kg

xyskwjy, 2021-11-25:
> 肯定修复源码撒，没有提pr 自己去改 改了提pr @82kg

也就是大雕库里的还是没修复。。。pr一下呗

WYC-2020, 2021-11-25:
我自己仓库改动太多，懒得pr 你自己拿去按我这个改了，你pr吧 ，还有一处改动在这里
[image]
我测试了一下没问题，你可以可以测试一下

WYC-2020, 2021-11-25:
@82kg

82kg, 2021-11-25:
> @82kg

> 谢谢大佬，明天测下试试

82kg, 2021-11-25:
> 我自己仓库改动太多，懒得pr 你自己拿去按我这个改了，你pr吧 ，还有一处改动在这里 [image] 我测试了一下没问题，你可以可以测试一下

谢谢大佬，明天测试下试试

82kg, 2021-11-26:
已经合并到master

closed 2021-11-26 by 82kg