coolsnowwolf/lede #8305: 网件R7800最新代码编译失败

hubbylei, 2021-11-25:
反馈bug/问题模板，提建议请删除

##  1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：R7800 全新编译

### (3) 详细日志

A：
Applying /home/hubbylei/lede_R7800/target/linux/ipq806x/patches-5.4/0065-arm-override-compiler-flags.patch using plaintext: 
patching file arch/arm/Makefile
Hunk #1 FAILED at 67.
1 out of 1 hunk FAILED -- saving rejects to file arch/arm/Makefile.rej
Patch failed!  Please fix /home/hubbylei/lede_R7800/target/linux/ipq806x/patches-5.4/0065-arm-override-compiler-flags.patch!
make[4]: *** [Makefile:38: /home/hubbylei/lede_R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.161/.prepared_0efc4d5fc902d5d168f527a100724841] Error 1
make[4]: Leaving directory '/home/hubbylei/lede_R7800/target/linux/ipq806x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/hubbylei/lede_R7800/target/linux'
time: target/linux/compile#7.56#27.25#24.32
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/hubbylei/lede_R7800'
make[1]: *** [target/Makefile:18: /home/hubbylei/lede_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/hubbylei/lede_R7800'
make: *** [/home/hubbylei/lede_R7800/include/toplevel.mk:230: world] Error 2

hubbylei, 2021-11-25:
Revert "kernel: bump 5.4 to 5.4.161 (#8294)"后编译成功

Jingwei-w, 2021-11-25:
> Revert "kernel: bump 5.4 to 5.4.161 (#8294)"后编译成功
我R7800回退到#8294这里还是编译报错，error和上面一样，环境啥的重新拉过一遍了

Jingwei-w, 2021-11-25:
> > Revert "kernel: bump 5.4 to 5.4.161 (#8294)"后编译成功
> > 我R7800回退到#8294这里还是编译报错，error和上面一样，环境啥的重新拉过一遍了

再往前回退，就能成功编译了，git reset --hard HEAD^了三次

hubbylei, 2021-11-25:
> > > Revert "kernel: bump 5.4 to 5.4.161 (#8294)"后编译成功
> > > 我R7800回退到#8294这里还是编译报错，error和上面一样，环境啥的重新拉过一遍了
> 
> 再往前回退，就能成功编译了，git reset --hard HEAD^了三次

我只把5.4.161内核的回退了就成功编译了,现在是5.4.159 -.-!

closed 2021-11-25 by coolsnowwolf