coolsnowwolf/lede #8312: 请问这个问题怎么解决？上次发过，issue关闭了也没搞清楚如何解决，见谅。

wangyang8284, 2021-11-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

Collecting package info: feeds/telephony/net/baresip
Collecting package info: feeds/telephony/net/freeswitch
Collecting package info: feeds/telephony/net/freeswitch-mod-bcg729
Collecting package info: feeds/telephony/net/kamailio
Collecting package info: feeds/telephony/net/miax
Collecting package info: feeds/telephony/net/pcapsipdump
Collecting package info: feeds/telephony/net/restund
Collecting package info: feeds/telephony/net/rtpengine
Collecting package info: feeds/telephony/net/rtpproxy
Collecting package info: feeds/telephony/net/sipgrep
Collecting package info: feeds/telephony/net/sipp
Collecting package info: feeds/telephony/net/siproxd
Collecting package info: feeds/telephony/net/sngrep
Collecting package info: feeds/telephony/net/yate
Collecting package info: merging...

Collecting package info: done
Collecting target info: merging...

Collecting target info: doneCreate index file './feeds/ssrp.index' 
find: ‘feeds/ssrp’: No such file or directory

Collecting package info: merging...

Collecting package info: donefind: ‘feeds/ssrp’: No such file or directory

Collecting target info: merging...

Collecting target info: done
Error: Process completed with exit code 1.

hubbylei, 2021-11-25:
"ssrp" 这个仓库clone成功了吗？

wangyang8284, 2021-11-26:
> 

Updating feed 'ssrp' from 'https://github.com//helloworld.git' ...
Cloning into './feeds/ssrp'...
fatal: could not read Username for 'https://github.com': No such device or address
failed.

果然没成功，感谢指导。怎么突然clone不下来了呢。。。晕死。。。 

感谢！

hubbylei, 2021-11-26:
地址错了，应该是https://github.com/fw876/helloworld.git

wangyang8284, 2021-11-26:
> 

感谢，真的是地址错了。

closed 2021-11-26 by wangyang8284